victronenergy/dynamic-ess #2: Provide pricing information for more countries

idstein, 2023-05-19:
I've been closely monitoring this development and we do plan to implement a dynamic ESS for Germany.
Currently, I can only call the API using country=NL, but I would really like to use EPEX Spot pricing for Germany to optimise the ESS system.

Is it possible to deploy a local backend service? We already have locally a home assistant solution running, recording all power consumption, pv production and running a PV forecast (via pvlib or solcast).

dirkjanfaber, 2023-05-30:
VRM fetches the day ahead pricing from Entso.eu. Which contains data from all (most?) European countries. The dropdown also shows all supported countries (including Germany).

Running the whole setup pure locally is not something that has been looked into yet.

idstein, 2023-05-30:
A simple test failed for me. Running the following yields a result only, if country=NL:
```
curl -vvv -X GET -G \
--url https://vrm-dynamic-ess-api.victronenergy.com \
-d site_id=XXXX \
-d b_max=14.5152 \
-d tb_max=3.36 \
-d fb_max=4.2 \
-d tg_max=15 \
-d fg_max=28.4 \
-d p_offset=0 \
-d b_cost=0.08 \
-d long=8.86429 \
-d lat=49.7205 \
-d country=NL \
--header 'X-Authorization: Token XXXX' \
--header 'Accept: application/json' \
--user-agent 'dynamic-ess'
```

Using `de` fails with the error: 
```
{"detail":"Could not get energy prices"}% 
```

dirkjanfaber, 2023-05-30:
On https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show there is no info on dynamic pricing. If I understand the naming correctly it is only available for the German speaking part of Luxembourg.

I am only familiar with the Dutch market, but do you have the possibility to have hourly tariffs in Germany? Which provider is that? In NL several providers offer an energy contract that has tariffs that change every hour. 

[image]

idstein, 2023-05-30:
For Germany there are dynamic day-ahead prices commonly used for larger feed-in installations (e.g. PV on farms). But also small PV installations are able to feed-in to the grid by dynamic day-ahead prices.
There exist a couple of tariffs for dynamic prices among others Tibber, Awattar and Voltego for small households. For larger business everyone basically has to apply for a dynamic business tariff that is based on the day-ahead prices plus a fixed rate determined by peak power consumption / feed-in in kW.

I'm not familiar with entsoe APIs, but by selecting DE+LU the data is shown. Likely, entsoe has some old market regions that combined DE+AT+LU or CH+DE+SK?

[image]

dirkjanfaber, 2023-05-30:
Entsoe provides the day ahead pricing on which all of the individual energy providers are basing their tariffs. The difference between providers is the amount of cents they add as margin per kWh. 

But your case is interesting; currently we only have Dutch testers using the system. I'll see if I can figure out why the German data is not showing.

idstein, 2023-05-30:
For feed-in energy there are mainly 3 available concepts in Germany (A) fixed tariff per kWh / EEG (B) day ahead spot price EPEX (minus a fixed  or profit share percentage) or (C) custom contracts with an energy broker.

For grid consumption a different tariff from the market is chosen. This is often a fixed price tariff (currently 30-40+ act / kWh) with a monthly baseline fee.
Consumers with more than 100 MWh per year are often in a dynamic tariff that is often influenced by spot price and a fixed monthly fee relative to the peak power consumption.

Taxation is for feed-in often not relevant (starts with 30+ kWp installations) for prosumers.

In case you require more information write me ;)

PoolsideKelbi, 2023-06-03:
Hey @idstein thanks for the detailed example of how the dynamic pricing is calculated in Germany. I did have a question about it though:
Looking at the simplified formula, which one of those factors are stale for the whole country and which ones would be dependant on the user/installation?

idstein, 2023-06-03:
Stale for the whole country would be spot price only, contributions/levy/taxes (except for a few hundred of the most energy hungry companies in Germany, but out of scope here) and VAT.
DSO working price depends on your location and who owns the distribution network.
As stated this only applies to grid consumption. For feed-in it is either a fixed rate, or spot + a fractional or fixed fee or EPP with custom agreements.

PoolsideKelbi, 2023-06-05:
Alrighty, thanks for the timely answer! 

If we look back at the formula of:
(Spot price * (1 + energy provider profit share) + DSO working price + Contributions/Levy/Taxes) * (1 + VAT)

It seems to me that the only percentage based additions here are "energy provider profit share" and VAT, the rest are all fixed fees that can be added up before VAT and after the provider fee. Just to double confirm, is this a generalisable formula for most dynamic energy contracts in Germany?

idstein, 2023-06-05:
Unfortunately, not true. Some use the formula mentioned (e.g. Awattar), but others (e.g. Voltego) use a fixed add-on fee per kWh.

So (A) e.g. Awattar
(Spot price * (1 + energy provider profit share) + DSO working price + Contributions/Levy/Taxes) * (1 + VAT)

or (B) e.g. Voltego
(Spot price + energy provider profit share in ct / kWh + DSO working price + Contributions/Levy/Taxes) * (1 + VAT)

dirkjanfaber, 2023-06-06:
After some discussions with the API developer, we agreed that allowing the user to fill out his/her own formula for buy and sell prices will be the quickest way to accomplish adding new countries. At a later stage we might be able to convert all different formula's back to a single one.

PoolsideKelbi, 2023-06-08:
@idstein, you can indeed leave out the p out of the formula and put the fixed price in. That means the price for consumption will be the same throughout the day.

dirkjanfaber, 2023-06-08:
This morning the `p` is no longer compulsory in the equation. So `0.405` should work too.