victronenergy/dynamic-ess #5: Feature request: Target amperage in via dynamic ess instead of max power point

sfstar, 2023-06-03:
Thank you for allowing users to start testing the new dynamic-ess feature.
Currently the node asks you to specify the max feedin/comsume (setpoints) in watts.
However, since the grid voltage can variate over the day I believe it would be better for users to be able to set a max amperage to target.
So that the charging amperage does not exceed the preferred amperage tolerance.
For example a desired amperage of 25A could mean 210*25 = 5250watt while 253*25 = 6325 watt.
This is more than a kwh difference that cannot safely be accomodated for by having to set a max wattage instead of amperage.
By targeting amperage it would allow the system to maximise the savings (while accounting for grid voltage) provided by this new (soon to be venus) feature.

dirkjanfaber, 2023-06-05:
Thanks for your feedback. Is your breaker failing because of this? There are ways within Node-RED to limit the maximum setpoint when the voltage is lower than expected.

dirkjanfaber, 2023-07-11:
Talked about this with the other developers. This is not something we are planning to do, as it will mean tracking two values (amperage and voltage) instead of just power.

closed 2023-07-11 by dirkjanfaber