victronenergy/dynamic-ess #9: Other graphs disappear after activating Dyn ESS

LecramRed, 2023-06-29:
When I activate the Dynamic ESS flow, other graphs in the Node Red dashboard disappear. And reverse: after disabling the Dyn ESS, the graphs appear again. What does the Dyn Ess flow do that other graphs disappear?

dirkjanfaber, 2023-06-30:
In order to generate the complex graphs, we use the `ui_template` node. Unfortunately this does bite with the original charts and so far we were unable to find a working way around this. As this node is a temporary solution (until the functionality moves to VRM) finding a way around this is not the highest priority. 

I did look into it and in theory it should be possible by moving away from the `ui_template` and putting all options in a `msg.ui_control` field (as shortly described [here](https://github.com/node-red/node-red-dashboard/blob/master/config-fields.md)). But that does not seem to work with adding plugins (which is used for graying out the past hours). Also putting text vertically on the y axis does not seem to be an option. If I look at the github repository of [node-red-dashboard](https://github.com/node-red/node-red-dashboard), it seems that under the hood a patched version 1.1.1 of chart.js is used. And that version is rather old. Our _arbitraryLine plugin_ node includes and uses version 4.3.0 instead. 

So unfortunately there is no quick solution to get this sorted. I'll leave this issue open for a few more days in case someone comes along with a working solution.

dirkjanfaber, 2023-09-12:
We've just published a new release, that is more incorporated into the Venus OS and follows the target SOC instead of setting a setpoint and a less than ideal way of idling the battery. This release is no longer dependent on the node-red-dashboard, so that also fixes this issue.

In order to use the new version, you will need to:
- update Venus OS (to 3.10 or above)
- update the victron-dynamic-ess node to version 0.1.1
- update the example flow
- reconfigure the Dynamic ESS node

More information can be found in the [release notes](https://github.com/victronenergy/dynamic-ess/blob/main/RELEASE-NOTES.md) and [README.md](https://github.com/victronenergy/dynamic-ess/blob/main/README.md)

closed 2023-09-12 by dirkjanfaber

LecramRed, 2023-09-12:
> We've just published a new release, that is more incorporated into the Venus OS and follows the target SOC instead of setting a setpoint and a less than ideal way of idling the battery. This release [...]

That is great, but the version in Node Red is still 0.0.43. Or am I too fast? https://flows.nodered.org/node/victron-dynamic-ess

dirkjanfaber, 2023-09-12:
Yes, you are a bit too fast. It is already [here](https://flows.nodered.org/node/victron-dynamic-ess). Node-RED usually takes a few hours to synchronize.