victronenergy/dynamic-ess #11: Does Dyn ESS only works with VE BMS?

Hhalewijn, 2023-07-03:
Customer is complaining he can not get the flow to work because of an error at node Victron-idle-battery. He's using 123SmartBMS.
Is it tied to only VE BMS?

dirkjanfaber, 2023-07-03:
The example flow is indeed tied to a VE BMS and sets the battery in idle mode by putting in a temporary charge schedule with the SOC set to minSOC. 

[image]

I am not familiar with the 123SmartBMS, but if you can somehow remotely steer the 123SmartBMS (or any other BMS) to become idle, this part of the flow needs to be changed. The "Battery Control" function sends out a `true` for setting the BMS to become idle and a `false` for becoming active again.

If you can't get it to steer via the dbus, perhaps it can be controlled via modbus or a web interface?

mpvader, 2023-07-03:
Hey @Hhalewijn and @dirkjanfaber , lets not use the word `VE BMS`, since there is no such thing. We have a BMS which is called the VE.Bus BMS, and another one called the `Lynx Smart BMS`, and yet more.

However: the POC Dynamic ESS flow, which is what this repo is about, works with all the other batteries as well. So Pylontech, MG, BYD, BMZ, Freedomwon, and so forth and so forth.

So: for some reason, something is different in the 123bms. Now, I know a bit how their integration works, so I can imagine something is different (it involves installing a script on the GX device, since 123BMS doesn't have a canbus port at the moment).

Its worthwhile to look into this from our side, at least a bit. To start that, please share the VRM url of that site in the slack channel @Hhalewijn , thanks! Then @dirkjanfaber can look into it.

mpvader, 2023-07-03:
ps. @Hhalewijn and can you add here in this issue a screenshot of the error?

Tussenbrock, 2023-07-07:
Goodmorning @mpvader I have te same problem as described here. I am using a Seplos Mason battery pack with a Seplos BMS. The BMS is communicating with the Victron system only not seen in the node red dynamic ESS.

dirkjanfaber, 2023-07-07:
@Tussenbrock Can you provide more information on the error and/or screenshots?

Tussenbrock, 2023-07-07:
[image]
If you need anything els let me know.

dirkjanfaber, 2023-07-07:
The message originates from [this piece of code](https://github.com/victronenergy/dynamic-ess/blob/fd1edfbce42e3bb85a18c88b206616b106e27e16/src/nodes/victron-idle-battery.js#L8). Which checks if there is a victron configuration client. 

Can you check if you have that available on Node-RED?

[image]

Tussenbrock, 2023-07-07:
Thanks for the quick reply @dirkjanfaber. I don't have the Victron Energy client.
[image]

dirkjanfaber, 2023-07-07:
This config client _should_ automatically be created. Question is why it does not happen on your site (and on Harolds customers site). I'd really like to take a look on your system to examine why it is not being created. 
Is it possible for you to share your site id with me so I can take a closer look? That can be done via mail, as you probably don't want to disclose that in a github issue. My mail address is dfaber at victronerergy.com.

Tussenbrock, 2023-07-07:
I have solved it by doing a new install of the node and now it all works, thanks for the great support!!

Tussenbrock, 2023-07-07:
[image]

dirkjanfaber, 2023-07-07:
Cool! Still wondering why the configuration node was not there in the first place, but glad that it is working now. 

@Hhalewijn: Can you check if a re-install of the node also works for your customer?

dirkjanfaber, 2023-07-07:
Took a look at the other site with the same problem. The quick-fix is to drag a Victron input node onto the canvas, select a service and input path (e.g. the sytem node, and selecting _serial_ or so). 

The _Victron Energy Client_ configuration node then appears on the configuration tab. 
[image]

I'll see if there is a way to have the dynamic ess node also create it, if it is not there yet.

dirkjanfaber, 2023-07-11:
Fixed in version 0.0.43.

closed 2023-07-11 by dirkjanfaber