victronenergy/dynamic-ess #12: Sweden not selectable as a country.

MissMusic, 2023-07-05:
Is there a reason why Sweden and Norway are missing in the country list?

danjul89, 2023-07-05:
Would also like Sweden, im using FI for now.

alvinpalv, 2023-07-06:
Waiting for the Swedish support! :)

dirkjanfaber, 2023-07-10:
We are using the data from https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show to get the needed day ahead prices. For some reason Sweden and Norway are split up into regions. That is why we skipped them for now (most other countries don't have those). 
Can we assume that each one of you knows which reason is needed for your connection? In which case we can add them as selectable options to the list.

danjul89, 2023-07-10:
> 

Alright, yeah most Swedes are based in SE3 and SE4(see picture below).
Im based in Sweden SE3 and @alvinpalv as well, you can perhaps create 4 "countries" named sweden i.e. **Sweden_SE[1-4]**

[image]

MissMusic, 2023-07-10:
>  Can we assume that each one of you knows which reason is needed for your connection? 

Yes, definitely. After the last few years with prices matching central Europe (quite a shock for us Swedes and Norwegians) everyone who didn't know what region they are in do now. If you make them as select-able country options it will be fine. If you want to be anal about it maybe rename the selection from country to energy region or something as that is what you are actually after.

dirkjanfaber, 2023-07-12:
Support for the extra regions has been implemented in version 0.0.43, which is available from the palette manager.

closed 2023-07-12 by dirkjanfaber