victronenergy/dynamic-ess #14: United Kingdom not available

apple047, 2023-07-07:
Hi, just came across dynamic-ess but UK is not available, would you consider supporting please?

Thanks

dirkjanfaber, 2023-07-10:
This will be solved via issue #12

dirkjanfaber, 2023-07-11:
I am sorry about that, but it appears that dynamic pricing for the UK is not provided by Entsoe right now. 

[image]

I will need to close the issue again as there is no way to add the UK without this information. Perhaps there is an alternative API available for the UK and we are just not aware of that. If you find one, please let us know and we'll add it.

closed 2023-07-11 by dirkjanfaber