victronenergy/dynamic-ess #15: Possible to manually select battery monitor?

MissMusic, 2023-07-08:
My system is using the Multiplus for battery monitoring and SOC calculations because the BMS is junk and unable to understand when the battery is full so it's reported SOC is just slowly drifting down to 0. I still have the BMS connected though, as it is feeding cell min/max and eventual fault info to the Victron system. The dynamic-ess is for some reason picking up SOC from the BMS when it searches for battery instead of selecting the same battery monitor as already selected for ESS in the system (in my case Multiplus-II).

Is it possible to add an option to manually select what battery monitor is being used by dynamic-ess, or change the battery identification to match what is already configured for the system?

mpvader, 2023-07-08:
Hi, how it *should* work is that dynamic ess uses the SOC of the system service on DBus.

which is also the one visible on main page in the gui, on VRM, and so forth.

You can define which source (multi, BMS, and so forth) is used for that in the Settings -> System setup menu.

can you check that as well as post a photo or screenshot of how you have that set up now?

MissMusic, 2023-07-08:
Here's what the node red flow fetches:
[image]

And here is how it is in the VRM portal. This SOC-value is also what is used when i configure ESS manually (like the minimum SOC unless grid fails setting)
[image]

mpvader, 2023-07-08:
Thank you. So, Victron is to solve your non-Victron crappy battery problems? 😉

More seriously: over to @dirkjanfaber for this. Above is all I can tell.

MissMusic, 2023-07-08:
@mpvader Yes! :D
To expand, I don't agree with Victrons mindset when interacting with BMS:es at all. Best case scenario it sort of works on a totally off-grid system with a BMS that for some unknown reason is designed to manage charging settings (to me that is not a job for the BMS at all). But as soon as you add ESS with grid feedback into the mix the whole idéa of relying on the BMS to control the MPPTs only out of a battery perspective it's just a bad design that obviously won't do what we want it to. Victron not having an override to disable or select what BMS features/settings are taking control of the system is a Victron problem, not a BMS one, as no BMS in the world could ever be expected to have features for grid-tied or feed in setups.

dirkjanfaber, 2023-07-11:
Fixed in version 0.0.43.

closed 2023-07-11 by dirkjanfaber