victronenergy/dynamic-ess #16: Node Red Dashboard UI not working

banne123, 2023-07-08:
Everything is activated as described, the battery is recognized in node red but on the UI all is blank. No graphs etc. What am I doing wrong?
[image]
[image]

banne123, 2023-07-08:
Also I'm getting this message
[image]

dirkjanfaber, 2023-07-09:
On the screenshot it looks like you disconnected and disabled the dynamic ess node. It should look like this instead:

[image]

banne123, 2023-07-09:
> On the screenshot it looks like you disconnected and disabled the dynamic ess node. It should look like this instead:
> 
> [image]
> On the bottom left of the edit panel there is the button to enable the node.
> 
> [image]
<img width="634" alt="Scherm­afbeelding 2023-07-09 om 18 49 21" src="https://github.com/victronenergy/dynamic-ess/a
[image]
ssets/139011357/6a8906e8-15ef-4c20-a1b1-524079159913">

Even when its enabled I get the same error on deployment. Also its not showing anything on the dashboard. There is a red triangle now at the graphs in the editor

dirkjanfaber, 2023-07-09:
You are missing the connection between the inject node and the dynamic ess node.

banne123, 2023-07-09:
> You are missing the connection between the inject node and the dynamic ess node.

How do I fix this? Ess id and API code are added

dirkjanfaber, 2023-07-09:
You can drag a connection from the end point of the inject node to the
input point of the dynamic ess node.

Then the red dots on the other nodes. You probably need to check in the
edit panel what is going on there. It should not be like that.
Perhaps the quickest solution is to re-import the flow from the examples
tab.

banne123, 2023-07-10:
> You can drag a connection from the end point of the inject node to the input point of the dynamic ess node. Then the red dots on the other nodes. You probably need to check in the edit panel what is [...]

With reimporting the red triangles are gone, also the ESS says ok now. Still its not working because of invalid nodes for the graphs
[image]
[image]

banne123, 2023-07-10:
Ok problem found, I made new flows everytime and disabled the old flows. Thats obviously not the way to work, after I removed all the "old" flows and created a completely new one the issue is solved