victronenergy/dynamic-ess #21: Feature Request: control Relay2 not only by the price but by current/expected PV Yield

khostri, 2023-07-11:
Is it possible to modify switch for Relay (2 preferably) to close and open not only based on negative price or below average price, but according to current energy balance. So if there is enough excess energy for heatpump or other usage.
Maybe in config node setup how much energy is needed for devices, controled by relay.
If there is enough excess for past 10 minutes for example and there is expected excess predicted, close relay. If excess drops and stays dropped for like 10 or less minutes and no better prediction, open relay.
I have never worked in node red and I started with that last week. But i have tried to do easy way for my heatpump using counter. (drop is faster then increase on purpose). But it can be done better with batch of last 10 minutes data + PV Yield estimation for example.
Adding my example which is active on my site currently together with Buderus (IVT) heatpump - relay 2 is connected to External input. external input is setup to PV system. When closed, Water comfort level is increased, temperature is also increased by 2 degrees in next checkpoint in queue.
[image]
[relay2.zip](https://github.com/victronenergy/dynamic-ess/files/12012877/relay2.zip)

dirkjanfaber, 2023-07-12:
One of the pitfalls is that this will work against the algorithms predictions and decisions. Which the existing example (switching on when the price is low) also does. At the moment we are not actively putting more effort into that.

As you are just starting with Node-RED, have you seen the examples on https://github.com/victronenergy/node-red-contrib-victron/wiki/Example-Flows? They might give you some ideas on how to do this.

khostri, 2023-07-12:
> One of the pitfalls is that this will work against the algorithms predictions and decisions. Which the existing example (switching on when the price is low) also does. At the moment we are not [...]

@dirkjanfaber Ok thanks. I just thought that for static installations it would make sense. Thats why I have attached my simple working example to your flow. just used output from negative price node to add additional nodes.
I am aware it can go against prediction and decision, but as I have mentioned in other comment I think you are doing recalculation every 10 or 15 minutes. I am not sure if recalculation is strictly only for future or if it takes past values to project for target values like SOC in the end of day/price period. I have though that in case of past values, consumption is known, expected and real PV Yield also and probably target endpoint for the day (or for the period of published dynamic prices). So based on configured values for node and putting real PV Y instead of predicted should let recalculation do it dynamically.
For the consumption change it is just normal operation even without relay(wash machine, laundry machine etc.) When relay closes, it will be some time consumption will be higher, so in that case algo can take it as new expected consumption level for defined timeframe (!5 minutes). Then recalculation starts again.

I have checked examples, but I was strugling with specifics: creating custom properties ie combine 2 or more messages into one, having time values back as array and remove oldest value each tick and recalculate average from rest + new one, etc.