victronenergy/dynamic-ess #22: Algo stops and starts at random moments

banne123, 2023-07-11:
I have this working for 2 cycles now and nothing the algo does or calculates makes sense to me. Its now 23:00, batteries are not empty, why does it stop discharging?? Lowest is at 30% highest at 54% SOC. It does not look at the SOC at all, it just stops because the algo says its time to stop. For charging the same issue, when time is time it stops charging. Besides that the charging time calculation makes no sense at all. Tomorrow will post another printscreen from the charging chart. Attached is the chart of today but look at the timeline at 23:00. Why is it stopping?
[image]

banne123, 2023-07-12:
Today same story. Why is it starting to charge at 10:00 and discharge after that and start charging again? This really makes no sense at all. End of story is that the battery is 79% charged at 17:00...?
[image]
[image]

banne123, 2023-07-12:
Same story for battery 2. What going wrong here?
[image]
[image]
[image]

PoolsideKelbi, 2023-07-12:
Hey @banne123, thanks for your comments and all the information you provided.

I am having a hard time understanding the concerns you have with regards to the algorithm's plan for the day. Perhaps if you could answer the following questions, I can be of help :)
- What is it about the current plan that does not make sense to you? 
- What do you expect the algorithm to do differently?
- What is the battery cost you have inputted to the algorithm?

banne123, 2023-07-12:
> Hey @banne123, thanks for your comments and all the information you provided.
>
> I am having a hard time understanding the concerns you have with regards to the algorithm's plan for the day. Perhaps [...]

Batterycost 1 cent, just to see what the algo would do with a "blank" price algorithm. 

What makes no sense I explained? It wants to charge today at 11:00 and discharge at 12:00, remain idle at 13:00 and then charge again till 17:00. What I expect the algo to do is calculate a time when its the best to start charging the battery and then charge the battery untill its fully charged

PoolsideKelbi, 2023-07-12:
Thanks for the timely response!

I see. The alternation between the charging and discharging is there because of the varying prices and the expected consumption. The algorithm does not have an inherent goal of charging or discharging the battery fully in one go. Its only goal is to minimise the costs made on the grid and battery. 

For instance, for the first installation screenshots you have provided, the reason why it charges at 11 is because it expects
consumption of around 4 kWh in the hours after and the FG_max is 3.5 kWh, meaning there is a need for battery to meet that 1 kWh in those hours. Hence it plans to discharge till 14. After 14 it starts charging at a high rate because the prices are the lowest at that time. It continues charging till 17, to be ready to meet the consumption needs in the evening where the prices will be more expensive. And then it discharges.

In doing these it never aims to charge fully, because that is not necessarily the most cost-effective way to go according to the algorithm. I hope this provides more context as to how the algorithm functions, if you have any more questions feel free to ask.

Note: looking at the screenshot you provided, it seems that you have NL dynamic pricing the graph of which should look roughly like this:
[image]

banne123, 2023-07-12:
> Thanks for the timely response!
>
> I see. The alternation between the charging and discharging is there because of the varying prices and the expected consumption. The algorithm does not have an [...]

I understand what you say but then i'm missing the point of the algo. I thought the algo was ment to trade with energy, charge batteries to the max on the lowest hours possible and then discharge at the most expensive hours. Charging to 79% or discharging during the day while the prices are cheap really does not make any sense to me at all. And yes i'm based in NL so looking at the NL graph for today I dont see any reason why it should discharge or remain idle at 12:00. 

Before this algo was installed I used a cronjob via SSH (for months already) for charging and discharging at static hours. That made more "profit" than how this algo does it

dirkjanfaber, 2023-09-12:
The target of the algorithm is not to maximize profit, but to minimize cost. See https://github.com/victronenergy/dynamic-ess#about.

closed 2023-09-12 by dirkjanfaber