victronenergy/dynamic-ess #23: Unexpected settings from dynamic-ess

MichaKersloot, 2023-07-11:
[image]

As a picture tells more than words. One would expect the grid setpoint to be set to 0 and battery idle to false when the proposal is 'from battery' as you can see at 1 o'clock.

andressade, 2023-07-12:
I see the same behavior. Even the calculated plan seems reasonable it will not follow it, only setting battery to idle and sitting there

PoolsideKelbi, 2023-07-12:
Hello @MichaKersloot, thank you for raising this matter.

The reason why the controls do not reflect what you see on the graph is because the system is trying to be conservative with allowing the battery to be idle. Since the forecasts for both consumption and solar yield can be inaccurate and the battery costs are a factor to consider, we have seen that it is best to disallow the battery to be non-idle when its planned usage is below 10% of its charge/discharge capabilities. 

I can provide a more detailed explanation over what happens here if you could provide me with the following parameters of your installation:
- B_max
- TB_max
- FB_max
- TG_max
- FG_max
Additionally, if you could provide me with the updated screenshot of the 'Energy graph', then we could see what really happened as a result of the decisions made by the algorithm for these early hours.

MichaKersloot, 2023-07-12:
Hi,

B_max:  9
TB_max: 3
FB_max: 3
TG_max: 4.1
FG_max: 4.1
Battery costs: 0.02

[image]

And for your convenience a battery soc graph of the same time:

[image]

There are some hiccups in the soc display because I missed: https://www.victronenergy.com/live/ccgx:root_access#partitions_read-only_rootfs_and_available_disk_space

It would helped me if this information was also on this page: 
https://www.victronenergy.com/live/venus-os:large

PoolsideKelbi, 2023-07-12:
Thanks for the timely reply!

It seems what I suspected is true:
For the hours 1 and 2, the planned battery discharge was below 0.3 (10% of the FB_max) looking at the screenshot you have provided in the original post. In this case the battery is set to be idle in order to be conservative with its usage. But for the hour 3, the planned discharge was above 0.3 hence the battery was not idle (as you can see from the screenshot you posted the last). 

It is important to note that the decision-making with regards to this conservative usage is made based on the forecasts not what happens in reality in that hour: for hour 2 the usage was actually above 0.3 while for hour 3 it was the opposite.

I hope this provided little more context why the strange behaviour exists when there is a small difference between the consumption and solar yield estimates.

MichaKersloot, 2023-07-12:
Thanks for your answer, I think I can follow the reasoning ;-) But I assume the settings and the graphics are from the same planning algorithm?

And for the conservative approach... I just see and it shows in the SOC graph that from 9 o'clock the planning is 'delivering to grid' the battery is not set at idle which makes my battery hover between 10% and 12%. Maybe setting the battery in 'idle' should solve this?

khostri, 2023-07-12:
> It is important to note that the decision-making with regards to this conservative usage is made based on the forecasts not what happens in reality in that hour: for hour 2 the usage was actually above 0.3 while for hour 3 it was the opposite.
>
Thank for Clarification @MichaKersloot / just curious, would help to use correction factor in recalculation periods? To take in account forecast-current delta. In each period its known if forecast was higher or lower than actual value and decision plan can be adjusted in such way.

MichaKersloot, 2023-07-12:
Hi @khostri , I'm not sure how the calculations work and how they should be adjusted. I just wondered why the actions/settings, based on the prediction don't match the graphics that should be based on the same predictions.

MichaKersloot, 2023-07-12:
Sorry to still bother you, but I try to understand how things work... 

At the moment it says:
[image]

Which is exactactly what is happening!
[image]

But how? Why is the battery not helping to achieve the requested setpoint? I can't find any setting or option that is limiting the battery from discharging.

MichaKersloot, 2023-07-13:
[image]

Some strange decisions again. Why charge my batter from the grid at the most expensive hours?

Is there some way to log the actual outcome that is send to the ESS functionality in the inverter?

Is there a (wiki/github) page where we can look into the internals of the decision making a bit? The comment about 10% backoff for example, was very helpful in understanding what happens.

PoolsideKelbi, 2023-07-13:
Hello again. I believe this was caused by an issue (overnight) on our end by the API. It has been resolved in the morning. Sorry about the outcome.

The issue was in the domain of overestimating the PV yield. So there is no sensical reason behind what happened overnight :)

MichaKersloot, 2023-07-13:
Ha, ehh... great... Thank you for clearing this up. I know this stuff is still beta? So I expect things like this can happen. Thank you for being open on this, so we can keep moving on ;-)

H8meCZ, 2023-08-06:
Is there any chance to set minimum SoC to same value - for example 20% (unless grid fails) for Dynamic ESS? It means the Dynamic ESS can change SoC from 20 to 100% not bellow in any chance. Thank you guys!

MichaKersloot, 2023-08-17:
> Is there any chance to set minimum SoC to some value - for example 20% (unless grid fails) for Dynamic ESS? It means the Dynamic ESS can change SoC from 20 to 100% not bellow in any chance. Thank you guys!

As far as I'm aware, it takes the settings from the ESS settings in the console: Settings -> ESS -> Minimum SOC (unless grid fails).

dirkjanfaber, 2023-09-12:
It does use the minimum SOC for its calculations. 

We've just published a new release, that is more incorporated into the Venus OS and follows the target SOC instead of setting a setpoint and a less than ideal way of idling the battery.

As this uses a different approach, you will need to:
- update Venus OS (to 3.10 or above)
- update the victron-dynamic-ess node to version 0.1.1
- update the example flow
- reconfigure the Dynamic ESS node (as we switched from the site ID to the VRM portal ID)

More information can be found in the [release notes](https://github.com/victronenergy/dynamic-ess/blob/main/RELEASE-NOTES.md) and [README.md](https://github.com/victronenergy/dynamic-ess/blob/main/README.md)

closed 2023-09-12 by dirkjanfaber