victronenergy/dynamic-ess #27: ESS doesn't discharge battery to grid

Mableeker, 2023-07-13:
Hi,

I've setup the ESS, charging goes well according to the calculated schedule.

Discharging is not going well, the ESS sets the setting "self-consumption above limit" to "PV" instead of "PV & Battery".

If I change it manually it starts to discharge, until the ESS resets the settings.

Furthermore, the discharging is limited to 300 Watts, while the limits are set to 2000 Watts. I can't find this eather.

Where can I modify this?

Regards, Marc

Mableeker, 2023-07-13:
[image]
[image]

Mableeker, 2023-07-13:
Yes it is with Node-red. It sets a schedule every 10 minutes and changes "self consumption above limit" to PV, so only PV is fed in the grid. If I select PV&Battery then it feeds in battery.

Mableeker, 2023-07-13:
According to the dashboard it should be feeding in right now. Grid setpoint is set to negative number (-2000), schedule is active with SOC limit 10%. The only setting that prevents feeding in is the "self consumption above limit"-> PV instead of PV&Battery....

I'm also still finding out, but I'm stuck at some points.

MichaKersloot, 2023-07-13:
What is the:
[image]

if that one is true, there is a schedule with a SOC setting and only from PV. My understanding is that when the 'set battery idle' is false no schedules should be enabled. 

Also just found out: Charging battery from solar means: Grid Setpoint: 0 , Set battery idle: false

Mableeker, 2023-07-13:
Rightnow it is feeding in (at the lowest price of the day...):
Grid setpoint (gs_min_max)
-0.6
Set battery idle
false
Enable Grid feed-in
true

I'll wait another 24 hrs tosee what happens and how the behaviour is.

MichaKersloot, 2023-07-13:
I suspect there is no active schedule anymore? 

I'm not sure how new your system is, but it seems some historical data in the VRM is needed ( I also only have 2 weeks of data now). I find the lower graph in the node-red dashboard with all the information (except prices) very informative. I'm not sure why the actions are not always the proposed actions, but that is a different issue ;-)

Mableeker, 2023-07-13:
Correct, no active schedule right now. I'll see how it works next 24 hrs. 

I still have to solve the issue that the maximum feedin is 300 watts. This is with and without node-red / ESS 

I think I have to revert the Venus OS on the MP II GX to a 2.x version via an USB stick, but the MP keeps teling me that there is no firmware. Reverting via remote VRM  didn't do the trick.

MichaKersloot, 2023-07-13:
As the MP II GX is the device that is inside the EasySolar II GX and I'm running V3.00 Large on my GX, I don't think a downgrade is necessary per sé.  There are Quite a lot settings that could influence the feed-in I think. Maybe you already know them, or you know more than I do, but I've found:
* On the MP II you could check the Input current limit as this is also the output limit
* There could be a BMS that is limiting the charge / discharge power (CCL / DCL)
* In the Settings -> ESS there is an option to Limit inverter power 
* In the Settings -> ESS -> Grid feed-in there is an option for Maximum feed-in
* In the Settings -> ESS there is an option for the Grid setpoint.

Mableeker, 2023-07-13:
Yes I know them all and checked them all.

There is something different, because it worked ok until I installed a home assistant integration on my ha server that fiddled with the Victron. Since then discharge is limited till 300 watts. Integration has been removed but issue persists.

Mableeker, 2023-07-13:
I tried that option, without result. I did veflash the firmware of the mp, without result. And some specific settings were not touched by these actions, what makes me think they are stored somewhere else, perhaps the gx section. That's why i want to revert the os.

Mableeker, 2023-07-13:
Thanks, didn't see this before, I'll try tomorrow!

Mableeker, 2023-07-14:
Result! That procedure did the trick. A lot of reconfiguration had to be done offcourse.... the hardest was to get dynamic Ess on node-red running again. I solved that by deleting the site in VRM and adding a new one.

Thanks for the help!

closed 2023-07-14 by Mableeker