victronenergy/dynamic-ess #32: [Bug]: Could not create a feasible schedule

andressade, 2023-07-18:
### VRM / Site iD

285851

### Country / region

Estonia (ee)

### B max

7.5

### TB max

2

### FB max

2

### TG max

0.72

### FG max

2

### Battery costs

0.04

### Buy price

(p+0.0044+0.0394)*1.2

### Sell price

p-0.0075

### feed-in possible

yes

### feed-in possible

yes

### What happened?

After enabling Dynamic ESS node the error "Could not create a feasible schedule" will be displayed. Api info output shows http 500 "Internal Server Error"

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

andressade, 2023-07-18:
[image]

dirkjanfaber, 2023-07-18:
Have you set the location in the VRM for the site?

Mableeker, 2023-07-19:
Same sort of situation here, small difference that in node-red flow it shows ok, but in the dashboard is says 'could not create a feasible schedule', although the graphs show a good schedule but the multiplus doesn't obey the schedule. Settings are not set...[image]

mrstackit, 2023-07-23:
I have the same issue in the flow and ui shows: Could not create a feasible schedule. I have do the initial Setup today and after a few hours this problem.

mpvader, 2023-07-31:
We're looking into this (Ali is)

PoolsideKelbi, 2023-08-01:
Hey @mrstackit, thanks for reporting this error.

This issue should be solved now. The low value of TG_max made it difficult for Dynamic ESS to schedule the rest of the day. But now it should be able to deal with values such as these.

Please give it a check and in the case that it works, feel free to close the issue :)

mrstackit, 2023-08-01:
Hey @PoolsideKelbi thanks for your reaction,

In the flow everything looks fine in the UI (dashboard the message is still there) even after restarting the flow. I activated the flow again at the weekend and just let it run. However, since 1 p.m. today, I have noticed strange behavior. The day was well planned until 1 p.m. Then it was somehow rescheduled and it suddenly started charging at 25 cents kWh, which made no sense. Also visible in the screenshots. No loading was planned there before 1 p.m. At 25 cents kWh it makes no sense to charge the battery if the most expensive price is around 30 cents. I am not the first person to report the bug, my settings in the flow are different:

Country: Germany
B max: 2.4
TB max: 1.2
FB max: 1.2
TG max: 0
FG max: 10

Should I open a new bug for this?

[image]
[image]

mrstackit, 2023-08-05:
Today again at 2pm fg_max is set at price 0,24 € kWh at SOC is on 90%. Thats bad. Prices in the evening max 0,32 € for two hours. Can you tell me why he suddenly sets fg_max even though it's not worth it? How is it calculated when the battery is charged from the grid?

closed 2023-09-12 by dirkjanfaber