victronenergy/dynamic-ess #33: [Feature]: Individual Grid Setpoint that is not Zero

mrstackit, 2023-07-23:
### Description

Hello, I think not all want to set the gird setpoint to zero. I think it's good that everyone can set their own grid setpoint and I would like this feature to be built in.

### Suggested Solution

-

mpvader, 2023-07-31:
Hi @mrstackit , right now the dynamic-ess algorithm is (ab-)using the grid-setpoint user setting because it had no other setpoint. This is being fixed right now as part of migrating Dynamic-ESS into VRM; no longer requiring Node-RED.

And most likely we'll also update the Node-RED implementation to use the new control points.

dirkjanfaber, 2023-09-12:
We've just published a new release, that is more incorporated into the Venus OS and follows the target SOC instead of setting a setpoint and a less than ideal way of idling the battery.

As this uses a different approach, you will need to:
- update Venus OS (to 3.10 or above)
- update the victron-dynamic-ess node to version 0.1.1
- update the example flow
- reconfigure the Dynamic ESS node (as it uses VRM portal ID instead of the Site ID)

More information can be found in the [release notes](https://github.com/victronenergy/dynamic-ess/blob/main/RELEASE-NOTES.md) and [README.md](https://github.com/victronenergy/dynamic-ess/blob/main/README.md)

closed 2023-09-12 by dirkjanfaber