victronenergy/dynamic-ess #35: According planning batter should be charged. But instead "idle"

SimonYoungtree, 2023-07-24:
### Contact Details

siempoot@gmail.com

### VRM / Site iD

287956

### Country / region

Netherlands (nl)

### B max

40

### TB max

6

### FB max

6

### TG max

10

### FG max

10

### Battery costs

0.04

### Buy price

(p+0.02+0.13)*1.21

### Sell price

(p-0.02)*1.21

### feed-in possible

yes

### feed-in possible

no

### What happened?

I can see in the Node Red dashboard that the system expects to charge the battery from 6pm. But instead it sets the battery to idle.
This way nothing can be charged.
[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

SimonYoungtree, 2023-07-24:
And at 6:58 PM it suddenly allowed the battery to charge. Really strange if you think about it. Prices are high around this time. It would have been smarter to charge during mid-day when prices were lower. To me it seems as if the system only calculates with the buying price and somehow for selling also takes the buying price and never takes the selling price into consideration.
In general, why buy from the grid while solar+battery during the day could have prevented this. I lose around 16ct (+/- 20- cent diff between buying and selling, minus batt costs) because of this behavior.

SimonYoungtree, 2023-07-24:
And a couple of minutes after 7pm the system sets the battery to idle again.

SimonYoungtree, 2023-07-27:
I have turned Dyn-ESS of for now

mpvader, 2023-07-31:
Hi @SimonYoungtree , we're working on an improvement, internally called "working with target SOC" which we expect will solve this. Maybe we'll add that into the Node-RED implementation, and for sure it will be part of the VRM implementation of Dynamic-ESS.

closed 2023-09-12 by dirkjanfaber