victronenergy/dynamic-ess #39: [Feature]: Input day ahead prices manually.

ajaxharg, 2023-07-29:
### Contact Details

ajaxharg@gmail.com

### Description

I have an E10 tariff in the UK.  This means that prices vary throughout the day in 6 distinct periods that run for different times. For me  cheap rate 0430-0730, 1330-1630,2030-0030. These actually run from 24 minutes past.  There is no way I could download this information for the day ahead.

### Suggested Solution

Allow manual times to be injected with the prices for periods. Also the allow entering pricing or not in my case for the export.

mpvader, 2023-07-31:
Hi, we've planned to look into the E10 / UK requirements in detail.

But only after we've finished what we're working on now, which is implementing dynamic-ess into VRM, so no longer requiring Node-RED.

dirkjanfaber, 2023-09-12:
While Matthijs is correct that the focus is on VRM right now, the implementation on Venus OS does allow using a Node-RED flow to set a schedule.

Basically set the mode to `4` (Node-RED) and insert your own schedule into the [slots](https://github.com/victronenergy/dynamic-ess/blob/main/RELEASE-NOTES.md#schedule-0-4).

You can use the [example flow](https://github.com/victronenergy/dynamic-ess/blob/main/examples/fetch-dynamic-ess.json) as a start. If you can get a working example or E10/UK, I am happy to add it to the importable example flows.

ajaxharg, 2023-11-04:
Hi, sorry for taking so long to reply. I've only just updated my cerbo to be able to look at the new node-red flows.  I think for someone wanting to use the schedule nodes they are a great addition and probably don't need an example as the one you have is pretty self explanatory.