victronenergy/dynamic-ess #40: [Bug]:

Tom-Moser, 2023-07-30:
### Contact Details

darktom@gmx.at

### VRM / Site iD

48e7da85adab

### Country / region

Austria (at)

### B max

14

### TB max

6

### FB max

7

### TG max

0

### FG max

3.6

### Battery costs

0.04

### Buy price

(p+0.02+0.13)*1.21

### feed-in possible

no

### feed-in possible

yes

### What happened?

Guten Tag 

Hab den Flow wie in der Anleitung installiert.  VRM Token erstellt, Tolle Anleitung übrigens.
nur der Baustein "VRM Dynamic ESS" hat ein rotes Quadrat [ object Object ] ich finde keinen Lösung ?
[image]
[image]

ich verwende keine Victron Batterie, kann das die Ursache sein ?  
[image]

ich hab alle meine flow´s deaktiviert um sicher zu gehen das die nicht stören. 
[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-08-09:
Thanks for the report. This should not happen. Are you still seeing this problem?

Tom-Moser, 2023-08-10:
Guten tag 
Hat sich nichts verändert. 
Eventuel liegt es an meiner Firewall. Wo kann ich sehen welche ports der baustein (node)  braucht, beziehungsweise auf welche seiten er zugreift?

dirkjanfaber, 2023-08-10:
The node does make a call to https://vrm-dynamic-ess-api.victronenergy.com. I'll try to replicate your issue next week to at least make the error more clear. Is there any more info in the debug tab that might help debugging this?

dirkjanfaber, 2023-09-01:
Sorry for the late response, but it looks like you are using the vrm_id (`48....`)  instead of the site id in the query. Yours is `14...`.

closed 2023-09-01 by dirkjanfaber