victronenergy/dynamic-ess #41: [Feature]: Plan for car charging

SimonYoungtree, 2023-08-03:
### Contact Details

siempoot@gmail.com

### Description

Dynamic-ESS has troubles foreseeing car charging. In fact it dumps the house battery during the evening and then at night, when I charge the car (23:30u and on) the house battery is empty and the charging is done from the grid. The house battery dump should have been done while charging the car.
I also charge the car sometimes during the day, when at home and the sun shines. There is no distinct rhythm for this.
Could you offer an approach for this?
- A link to my car, perhaps together with Jedlix?
- or, provide an advise (via vrm, app etc) about when to charge the car, for me to manually adapt the car charging schedule.
- or, would it be better to always charge during the same time in the night, assuming the dyn-ess algorithm will learn this. 
- or any other advise?

(FYI I have about 40kwh nett capacity in the house battery)

### Suggested Solution

Could you offer an approach for this?
- A link to my car, perhaps together with Jedlix?
- or, provide an advise (via vrm, app etc) about when to charge the car, for me to manually adapt the car charging schedule.
- or, would it be better to always charge during the same time in the night, assuming the dyn-ess algorithm will learn this. 
- or any other advise?

### Alternatives

I have no insight in the learning capabilities of the dyn-ess algorithm. 
I could imagine that dumping the battery is postponed until the expensive hours in the morning (6-9am, or maybe longer) and never done during evening hours. In that case dyn-ess should offer an option to allow dumping to the grid only starting at 6am (only if room needs to be  provided to catch the harvest from the sun during the day)

### Additional Context

I have 40kwh net storage in the house battery. Typically charging the car will take 20-40 kwh. Sometimes more.
my house consumes 8-10 kwh daily, but this can double in winter

LecramRed, 2023-08-21:
I have the same issue. I am now charging my car while the energy price is low. In stead of charging the home battery and car battery from the grid, it is unloading my home battery into my car. In this way, there is no energy left at the end of the afternoon, so the dump at 19-21 hours, when the energy price is high, will not succeed.

mdkeil, 2023-08-31:
> In stead of charging the home battery and car battery from the grid, it is unloading my home battery into my car.

for this case, you can (auto)set the victron to "charger mode" to prevent charging from battery to car.. or you can (auto) limit the discharge power for this time period.

SimonYoungtree, 2023-08-31:
Yes, you can tweak the system, but this is not optimal. An algorithm to determine when to optimally charge, discharge, freeze battery, etc., can only take things into account from which it "knows" what is going to happen. This can be from history (usage), or from the future (pricing, amount of sunshine), but a car charging event that happens sometimes during the night, sometimes during the day, or sometime not all is such a disturbing big event, that an algorithm cannot cope with it. You need some means to let it know ahead that the event will take place at some specific time.

LecramRed, 2023-08-31:
I solved it for now with a flow in Node Red. When the current through the car charger exceeds 1000 Watts, the inverter power is set to zero. And back when the car charger current drops. 
[image]

LecramRed, 2023-09-01:
May be this works: put the installation at the critical load, except the car charger. Than set this setting; 
[image]

SamuelBrucksch, 2023-11-07:
The issue is not only, that the battery is drained by charging the car, but also that the prediction changes. I also randomly charge the car, so it probably is not really possible to estimate it. So there should be a way to tell the system that the car is charging and that it should not count that to the consumption prediction algorithm.. I can for example tell Victron with how much power the car is charging, so it can be substracted from the current power readings.

mman, 2023-11-10:
@SamuelBrucksch I have a custom node-red flow that reduces Multiplus Max Inverter Power to 90% of what the EV charger is pulling every second and that way I can guarantee that the battery is essentially left alone when the EV charger is active. Agree 100% with you that EV car charging has to be treated very specifically in the Victron world (and all other home PV systems FWIW). I would even go that far that it has to be treated as a special entry on the "consumption" charts because it really negatively affects the consumption forecast and breaks everything else... But I'm sure fine folks at Victron are all aware of that...

vbdaniel4, 2024-02-13:
Solution could be: extra meter (or input via mqqt or something) for the car wall box.
Then dynamic ess can handle forecast better.
Dynamic ess could decide if price is high/low, therefore decide if it should use battery capacity for charging the car (now), or keep power for later (evening, early morning)...

aterjung, 2024-05-27:
Hi,
i did a look in the source to see if i could do anything to remove my car Charger from the forcast. Unfortunately, it seems to happen all in VRM cloud. I have not found andy code that submits meterdata or the like in this code. I have an extra meter for my WB and could easily remove the Chargers consumption from my Grid-Meter data cause a allready suply it via mqtt. But in this case i will lose the ability to charge the car from the PV Panels. Is there anything new on this topic?

johnsonhdw2, 2024-08-20:
same here, dynamic ess is working good but the forecast is being messed up when charging the car on different times of the day.
I could easily subtract the charger consumption but then I cannot charge from PV/ Victron does not have the correct overall consumption data.
It would be really nice to remove the car charging completely from dyn ess, since this better being done with something like evcc.