victronenergy/dynamic-ess #42: [Bug]: "need a victron config client" if this is the very first victron flow added to node-red

trooster, 2023-08-09:
### Contact Details

jjtrooster@gmail.com

### VRM / Site iD

246808

### Country / region

Netherlands (nl)

### B max

19.2

### TB max

12

### FB max

12

### TG max

17.25

### FG max

17.25

### Battery costs

0.02

### Buy price

(p+0.1435)*1.21

### Sell price

(p+0.1435)*1.21

### feed-in possible

yes

### feed-in possible

yes

### What happened?

If this is the very first flow added to node red, the _idle battery_ flow gives an error on the _victron-idle-battery_ node:

"need a victron config client"

The Victron config client is only available when there is a Victron energy flow created before. After I added a flow with one of the normal Victron Energy nodes, the config client was created and the dynamic ESS flow started to work.

I don't know if this can be fixed in the ESS "victron-idle-battery" node, or that this must be added to the prerequisites in the documentation.

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-08-10:
Thanks for reporting. Just to make sure, you are running the latest version? IIRC I've fixed this with [this](https://github.com/victronenergy/dynamic-ess/commit/f389ea40cd6209bab23e0fa240a3e436a2d99743) commit, but I'll take a closer look at it next week. Might be that I did not include that fix in the latest release (or that it does not work all of the time).

trooster, 2023-08-17:
Hi Dirkjan, I downloaded the one that was available in the latest Venus OS Large at the time I wrote the report.

dirkjanfaber, 2023-09-01:
For the newer flow we remove the victron-idle-battery node, so this will no longer be an issue.
There is still a bit of development time needed before that one will be released though.
So closing this issue as won't fix. Work-a-round is to deploy another Victron Node first and then use the example flow.

closed 2023-09-01 by dirkjanfaber