victronenergy/dynamic-ess #43: Why doesn’t Victron obey nodered schedule

Mableeker, 2023-08-19:
I’ve installed Venus os large, node red, dynamic ESS. Charging according to schedule goes well, but discharge not. It sets to pv only not pv & battery. Again… why is this only with me when everyone is fine? Struggling with it for 3 months now ..
[image]
[image]
[image]
[image]
[image]

Mableeker, 2023-08-19:
Disabled again unfortunately still not working here...

Mableeker, 2023-08-19:
Scheduled charging goes well, discharging not

MichaKersloot, 2023-08-20:
Hi,

Your screenshot shows the battery is set to idle, so that's the reason for the settings in the GX device.  The reason why, is something the portal decides. But costs and minimum battery usage is taken in consideration.

Mableeker, 2023-08-20:
At this moment 21.20 hrs, it should be discharging, but unfortunately no action.

Mableeker, 2023-08-20:
[image]

Mableeker, 2023-08-20:
[image]

MichaKersloot, 2023-08-20:
And what were the ESS settings at that time? There shouldn't be an active schedule in de ESS settings and the setpoint should be at -3000 as far as i'm aware.

Mableeker, 2023-08-21:
I'll check again this evening, the settings were as shown in the posted screenshots. So an active schedule.

Mableeker, 2023-08-21:
Right now it should be discharging, but nothing happens, it has an active schedule [image][image]

MichaKersloot, 2023-08-21:
Hi,

i'm sorry, but there is just one more or less important piece missing in your screenshots. And that's the bottom line which shows the actual status of the system. I'm interested if the battery is set to idle when there is an active schedule. This way we can check if the node-red controls and the GX device are in sync.

If that is the case, the next thing would be to find out why the graphs and the controls are out of sync.

Mableeker, 2023-08-21:
This one?
[image]

MichaKersloot, 2023-08-21:
Yes, and the check there is no active schedule at 21:xx in the GX device. The screenshots above are from 20:xx

MichaKersloot, 2023-08-21:
Hmm.. not sure if it matters, but looking at your screenshots you've set mode to: 'optimized (with batterylife)'. Because of my LiFepo4 batteries I've chosen for 'optimized (without batterylife)'.

Mableeker, 2023-08-21:
I'll change that and see what happens today 👍. I have pylontech us5000 connected.

Mableeker, 2023-08-21:
It's discharging 👍
[image]
[image]
[image]

Mableeker, 2023-08-21:
[image]

Mableeker, 2023-08-23:
It's working for 3 days now, according to charge/discharge schedule. The setting 'optimized with/without battery life' seems to make te difference. Thanks for the help Micha!

closed 2023-08-23 by Mableeker