victronenergy/dynamic-ess #44: [Feature]: Dynamic input variables

Gr1pen, 2023-08-29:
### Contact Details

tonny@gripagarden.se

### Description

As it is now the parameters for To Grid, From Grid, To Battery, From Battery etc is configured as static values in the configuration of the node. If these values could be configured by sending messages to the input of the node, it would be really easy to adjust to different installations and needs and also easy solve some of the other feature requests.

Look at the "Power Saver" node and the "dynamic config":
https://powersaver.no/nodes/ps-strategy-lowest-price.html#description
### 

### Suggested Solution

Make the node accept incoming messages to control the node's config parameters dynamically.

dirkjanfaber, 2023-08-29:
It already does that. You can set different `msg` fields, e.g. inject a `msg.tb_max`. 

[image]

closed 2023-08-29 by dirkjanfaber

Gr1pen, 2023-08-29:
That is just fantastic! :)  
Sorry I missed that in the documentation.