victronenergy/dynamic-ess #48: [Bug]: last hour of day (23) from grid but never from battery eventhough there should be enough in battery

trantoriana, 2023-09-06:
### Contact Details

victron@ophthalmophobia.net

### VRM / Site iD

314806

### Country / region

Austria (at)

### Sell price

0.13

### feed-in possible

None

### feed-in possible

None

### What happened?

Looks like the last hour of the day always imports from grid rather than from battery as the other hours do nicely.

### Screenshots

```bash
![plan](https://github.com/victronenergy/dynamic-ess/assets/24240995/508b0f09-1044-467c-9c85-e8f447904711)
```

trantoriana, 2023-09-06:
[image]

dirkjanfaber, 2023-09-06:
At the end of the day, the new prices of tomorrow are already used by the algorithm, It starts using that info as soon as it gets it (around 15.00). So it is very well possible that the algorithm decides to charge for cheap prices and start discharging further in the morning of the next day. 
Looking at the logs of your site, it shows that it definitely does not _always_ use the grid the last hour of the day.

trantoriana, 2023-09-06:
Hi Dirkjan,

I disagree closing the ticket because the logs do not confirm 'always'.

I am not convinced by your statement that the algorithm will charge later again, that is an assumption the graph does not show as it is not a rolling graph, that is another topic. At least the graphs of the next day does not show that to me.

[image]
[image]

(Had to do that on mobile, sorry for the small sizes).

The algorithm prefers the grid the whole night, then dumps a bunch of power on the grid.. that does not make sense to me at all. The battery cost per kWh is around 4 cent, which is way less that the grid price during the whole night, though grid is lower than dumping over the same period.

Please be mindful that my settings are a constant 13 cents price for dumping energy on the grid... I would kind of understand it when the dumping price would also relate to 'p', but it does not currently.
(I had some tests done with dumping price be 80% of p, perhaps that is still used somehow..).

Also, I only use the graphs for learning now, so the actual grid point is never set, perhaps that is messing up the decisions as the actuals are way different than the plan?

Is there some documentation about the algorithm itself that I can read so I may understand the decisions better?

trantoriana, 2023-09-06:
And finally, also today's 23 hours seems to behave the same...

[image]

It does not sit right with me.
Perhaps I just do not understand the algorithm but that makes it difficult to verify whether it is useful to actual enable.