victronenergy/dynamic-ess #49: [Feature]: Rolling graphs rather than fixed for a single day until 00:00

trantoriana, 2023-09-06:
### Description

Currently the graphs are day bound meaning the end at midnight and show progress during the days as grey areas.
Unfortunately these graps are always day based, so looking at it around 23:00 does not show any plans for after midnight.

### Suggested Solution

Rolling graphs with 24 forecast rather than fixed for a single day until 00:00.

A better understanding of the generated plan would be a view roll in the hours of the next day including plan for usage and battery soc.
This would give much more confidence that battery SOC over the night is sufficient the cover the expected load until the sun rises.

The way I envision something like this is to have the graps having a slightly larger view with some hours visible in the past and always have the next 24 hours as 'planned'.
With each passing hour one hour from the past drops off from the left and one hour in the future is added on the right.

Expected usage, battery soc, PV forecast, loading schedules based on hourly prices etc.... can be rolled in as they become available.

With this when ever you look will have a 24 hour plans, even at 23:00 at night...

dirkjanfaber, 2023-09-06:
Thanks for the suggestion and agreed that it would be really nice. Looking always 24 hours ahead is unfortunately not possible (new energy prices only get published around 15.00). Which makes it hard to capture in an always functional graph. So for the Node-RED implementation adding this probably won't happen.
That being said, the implementation in VRM is a lot more likely to get that. Though i cannot guarantee that will be already in the first release either.

closed 2023-09-06 by dirkjanfaber

dirkjanfaber, 2023-09-15:
To add to this, the new version has 48 hour graphs in (as soon as the new prices get in).