victronenergy/dynamic-ess #50: [Feature]: Fixed price Economy 7 in UK

andrewdavidson64, 2023-09-07:
### Description

Support for fixed price Economy 7 for UK customers.

### Suggested Solution

I have a fixed price tarrif that runs between 1am and 8am UTC, where those 7 hours are cheaper, and the other 17 hours are less cheaper.  I don't need any calculations regarding 'p' other than the fixed price.

Also, I'm having issues setting the system up as I run my Cerbo on UTC-7mins (my electricity meter has a fixed inbuilt clock to switch tarrifs, and it's 7 minutes slow from factory!).  There is no UTC selection under the timezone/location setting in the NODE-Red setup options.

MichaKersloot, 2023-09-08:
Think you better not use this Dynamic ESS feature as it is for dynamic pricing. It looks like you could be able to achieve this functionality with the default ESS functionality by setting scheduled charge levels. 
See more on: https://www.victronenergy.com/media/pg/Energy_Storage_System/en/configuration.html

dirkjanfaber, 2023-09-22:
You should be able to fill fixed prices out in the new beta-VRM implementation. See the [community post](https://community.victronenergy.com/articles/232720/dynamic-ess-on-beta-vrm.html) for more info on that.