victronenergy/dynamic-ess #51: [Bug]:

Marous23, 2023-09-12:
### Contact Details

marek.vagner@gmail.com

### VRM / Site iD

318034

### Country / region

Czech Republic (cz)

### B max

12

### TB max

12

### FB max

12

### TG max

12

### FG max

12

### Battery costs

0.04

### Buy price

p

### Sell price

p

### feed-in possible

yes

### feed-in possible

yes

### What happened?

Hi!
first of all thanks for this amazing project. I just started to use that in my home appliance. I see a problem in timezone that I decided to report. In Czechia we are switching between CET (UTC+1) and CEST (UTC+2). Right now we are still in CEST.
The CET/CEST change is done in exact dates, March 26 and October 29.
The problem is currently 1h before. For example today I see the price peaks at 7am and 7pm. In real it should be in 8am and 8pm. The price differs significantly.
Here is visible the difference:
https://www.ote-cr.cz/cs/kratkodobe-trhy/elektrina/denni-trh?date=2023-09-12

https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show?name=&defaultValue=true&viewType=GRAPH&areaType=BZN&atch=false&dateTime.dateTime=12.09.2023+00:00|CET|DAY&biddingZone.values=CTY|10YCZ-CEPS-----N!BZN|10YCZ-CEPS-----N&resolution.values=PT15M&resolution.values=PT30M&resolution.values=PT60M&dateTime.timezone=CET_CEST&dateTime.timezone_input=CET+(UTC+1)+/+CEST+(UTC+2)

[image]

[image]

Thank you.

Marous23, 2023-09-12:
Adding one more screenshot to show that Dynamic ESS is working with 1h different price list that it should. It should consider the CEST time in Czechia to have a peak in 8am instead of 7am.
[image]

khostri, 2023-09-12:
Hi,
I think its correct. Prices you are seeing in OTE are labeled differently - hour 1 means 0-1. (ie first hour of the day) so hour 8 means 7-8 which is correct.
In NodeREd you see hour labeled 7 which means more logically 7-8 (ie hour starting at 7).

Marous23, 2023-09-12:
Hi!
Oh, sorry. I missed that. Now I see it.
ENTSO-E is using number row 0-23 for hours.
OTE 1-24.
That was source of my confusion that 8 is not 8 :)

In that case my bug report does not make sense. There is no problem at the end. 

I am sorry again and many thanks for the clarification.