victronenergy/dynamic-ess #52: [Bug]: Spanish market price is not correct

TomasVelPar, 2023-09-12:
### Contact Details

tomas.velasco.parra@gmail.com

### VRM / Site iD

79108

### Country / region

Spain (es)

### B max

24

### TB max

2.7

### FB max

5

### TG max

0

### FG max

3.5

### Battery costs

0.04166667

### Buy price

p

### Sell price

p

### feed-in possible

yes

### feed-in possible

yes

### What happened?

at least buy prices at Spain are not obtained correctly. 
For example the real prices (sell and buy) are the following ones (for the peninsula):

[image]

they are obtained from: 

https://www.esios.ree.es/es/analisis/1739?compare_indicators=1001&vis=1&start_date=12-09-2023T00%3A00&end_date=12-09-2023T23%3A55&compare_start_date=11-09-2023T00%3A00&groupby=hour

and the dynamic ESS shows this other ones (the same for buy and sell):

[image]

To obtain the real prices of buying (5 diferent types in Spain (Peninsula, Baleares, Canarias, Ceuta and Melilla)) and selling it could be used the api of the oficial api web: https://api.esios.ree.es/

For example i use the following web for the 5 buy prices: 

url = 'https://api.esios.ree.es/indicators/1001?start_date=' + day_1 + 'T00%3A00%3A00&end_date=' + day_2 + 'T23%3A59%3A59'

and for the sell prices this other one:

url = 'https://api.esios.ree.es/indicators/1739?start_date=' + day_1 + 'T00%3A00%3A00&end_date=' + day_2 + 'T23%3A59%3A59'

Thanks so much!

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-13:
Thanks for the clarification. To be honest, adding this api to the system will probably take a while. Getting all of the bugs out and everything stable for the areas that follow the ENSTSOE prices is on top of the priority list for now.

TomasVelPar, 2023-09-13:
Thanks so much again. So, until you have time to deploy this, is there any way (for users) to send the hourly price of buying and selling to the nodered or to somewhere to let the dynamic ESS still working?

dirkjanfaber, 2023-09-13:
Not yet, but allowing the users to provide their own hourly prices for the calculations is definitely a good idea.