victronenergy/dynamic-ess #53: [Bug]: Sell and Buy prices don't change in the graph v 0.1.1

LecramRed, 2023-09-12:
### Contact Details

info@redupsell.nl

### VRM / Site iD

345657

### Country / region

Netherlands (nl)

### B max

12

### TB max

3.5

### FB max

4.5

### TG max

15

### FG max

15

### Battery costs

0

### Buy price

(p+0.02+0.13)*1.21

### Sell price

(p+0.02+0.13)*1.21

### feed-in possible

yes

### feed-in possible

yes

### What happened?

I love the new 0.1.1 update. But when I change the price rules in the DESS node, and the battery costs, nothing changes in the graph. The gap between Buy and Sell remain the same.

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

LecramRed, 2023-09-12:
I removed both dess and dess config flows, but it still shows the wrong prices. Also still the wrong formula in the object in the newly generated dess config flow

LecramRed, 2023-09-12:
I had too restart Node Red, too. Now it shows the right graph

dirkjanfaber, 2023-09-13:
Can you add the following to your flow?
```
[
    {
        "id": "11c0695b7819894f",
        "type": "group",
        "z": "89827d782ab51de2",
        "name": "Store battery capacity",
        "style": {
            "label": true
        },
        "nodes": [
            "e9a9bebc9c4732d0",
            "b48a8f507e14330a",
            "b68ffdbbf4152cf3",
            "8d723ea6a3f644bc"
        ],
        "x": 54,
        "y": 819,
        "w": 1092,
        "h": 82
    },
    {
        "id": "e9a9bebc9c4732d0",
        "type": "change",
        "z": "89827d782ab51de2",
        "g": "11c0695b7819894f",
        "name": "battery capacity",
        "rules": [
            {
                "t": "set",
                "p": "payload",
                "pt": "msg",
                "to": "dess.options.b_max",
                "tot": "flow"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "x": 260,
        "y": 860,
        "wires": [
            [
                "b68ffdbbf4152cf3"
            ]
        ]
    },
    {
        "id": "b48a8f507e14330a",
        "type": "victron-output-custom",
        "z": "89827d782ab51de2",
        "g": "11c0695b7819894f",
        "service": "com.victronenergy.settings",
        "path": "/Settings/DynamicEss/BatteryCapacity",
        "serviceObj": {
            "service": "com.victronenergy.settings",
            "name": "com.victronenergy.settings"
        },
        "pathObj": {
            "path": "/Settings/DynamicEss/BatteryCapacity",
            "name": "/Settings/DynamicEss/BatteryCapacity",
            "type": "number"
        },
        "name": "",
        "onlyChanges": false,
        "x": 880,
        "y": 860,
        "wires": []
    },
    {
        "id": "b68ffdbbf4152cf3",
        "type": "change",
        "z": "89827d782ab51de2",
        "g": "11c0695b7819894f",
        "name": "to number",
        "rules": [
            {
                "t": "set",
                "p": "payload",
                "pt": "msg",
                "to": "$number(payload)",
                "tot": "jsonata"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "x": 450,
        "y": 860,
        "wires": [
            [
                "b48a8f507e14330a"
            ]
        ]
    },
    {
        "id": "8d723ea6a3f644bc",
        "type": "link in",
        "z": "89827d782ab51de2",
        "g": "11c0695b7819894f",
        "name": "link in 1",
        "links": [
            "0a258a4a88409834"
        ],
        "x": 95,
        "y": 860,
        "wires": [
            [
                "e9a9bebc9c4732d0"
            ]
        ]
    }
]
```

dirkjanfaber, 2023-09-13:
This should be fixed in the 0.1.2 update.

closed 2023-09-13 by dirkjanfaber