victronenergy/dynamic-ess #54: [Bug]: ver 0.1.1: actual discharging way too low to meet planned SOC level

SimonYoungtree, 2023-09-12:
### Contact Details

siempoot@gmail.com

### VRM / Site iD

c0619ab31cc4

### Country / region

Netherlands (nl)

### B max

40

### TB max

5

### FB max

7

### TG max

16

### FG max

16

### Battery costs

0.027

### Buy price

(p+0.13)*1.21+0.02

### Sell price

(p+0.13)*1.21+0.02

### feed-in possible

yes

### feed-in possible

yes

### What happened?

After enabling the new 0.1.1 version, the schedule planned to lower the battery with 15% in one hour (about 6kwh). To reach this I would expect the Multiplus (3x) would have to do some serious work. However it took  around 300 to 400 watt from the battery. Also next hour, also with a plan to lower with 15%, exactly the same. 

And after 9pm, it still has the plan to lower the SOC, but instead it is charging with around 300 watt from the grid.

I have no errors and the install went fine. What is the problem?

[image]
[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

LecramRed, 2023-09-12:
I had that too. After deleting the dess flow and restarting Node Red, the discharge current has increased a lot

dirkjanfaber, 2023-09-12:
At the moment your target SOC is set to 31% with the battery SOC at 31.2%

[image]

Please allow the system to run for a while. We'll take a closer look when needed.

LecramRed, 2023-09-13:
Target is now 56%, actual soc 42%. Charge power is now 220W. Isn't that way too low for a 12 kwh battery?

LecramRed, 2023-09-13:
Last hour, the target was 69%, but the current was way too low to achieve this and reached only less than 50 at the end of the hour. It's target is now 57%, 15 minutes to go, 50% soc and 200 W charge power. Something is going wrong.

LecramRed, 2023-09-13:
[image]
[image]

Target is 11%, estimated discharge is about 3.5 kWh, but the discharge power is about 800W. It will not achieve it's soc target in this way.

dirkjanfaber, 2023-09-13:
Totally agreeing that it looks like there is a problem right now. Investigating the cause.

LecramRed, 2023-09-13:
Together we make it great!

dirkjanfaber, 2023-09-13:
Can you add the following to your flow?
```
[
    {
        "id": "11c0695b7819894f",
        "type": "group",
        "z": "89827d782ab51de2",
        "name": "Store battery capacity",
        "style": {
            "label": true
        },
        "nodes": [
            "e9a9bebc9c4732d0",
            "b48a8f507e14330a",
            "b68ffdbbf4152cf3",
            "8d723ea6a3f644bc"
        ],
        "x": 54,
        "y": 819,
        "w": 1092,
        "h": 82
    },
    {
        "id": "e9a9bebc9c4732d0",
        "type": "change",
        "z": "89827d782ab51de2",
        "g": "11c0695b7819894f",
        "name": "battery capacity",
        "rules": [
            {
                "t": "set",
                "p": "payload",
                "pt": "msg",
                "to": "dess.options.b_max",
                "tot": "flow"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "x": 260,
        "y": 860,
        "wires": [
            [
                "b68ffdbbf4152cf3"
            ]
        ]
    },
    {
        "id": "b48a8f507e14330a",
        "type": "victron-output-custom",
        "z": "89827d782ab51de2",
        "g": "11c0695b7819894f",
        "service": "com.victronenergy.settings",
        "path": "/Settings/DynamicEss/BatteryCapacity",
        "serviceObj": {
            "service": "com.victronenergy.settings",
            "name": "com.victronenergy.settings"
        },
        "pathObj": {
            "path": "/Settings/DynamicEss/BatteryCapacity",
            "name": "/Settings/DynamicEss/BatteryCapacity",
            "type": "number"
        },
        "name": "",
        "onlyChanges": false,
        "x": 880,
        "y": 860,
        "wires": []
    },
    {
        "id": "b68ffdbbf4152cf3",
        "type": "change",
        "z": "89827d782ab51de2",
        "g": "11c0695b7819894f",
        "name": "to number",
        "rules": [
            {
                "t": "set",
                "p": "payload",
                "pt": "msg",
                "to": "$number(payload)",
                "tot": "jsonata"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "x": 450,
        "y": 860,
        "wires": [
            [
                "b48a8f507e14330a"
            ]
        ]
    },
    {
        "id": "8d723ea6a3f644bc",
        "type": "link in",
        "z": "89827d782ab51de2",
        "g": "11c0695b7819894f",
        "name": "link in 1",
        "links": [
            "0a258a4a88409834"
        ],
        "x": 95,
        "y": 860,
        "wires": [
            [
                "e9a9bebc9c4732d0"
            ]
        ]
    }
]
```

LecramRed, 2023-09-13:
I would love to, but would you please tell me how to add and where?

dirkjanfaber, 2023-09-13:
Sure. You can import this into Node-RED via the ctrl-i / command-i / via the menu and then pasting it into the text area.

[image]

[image]

LecramRed, 2023-09-13:
Thanks, I have done it. What should it do? Where do I have to pay attention to, so that I can inform you about the results?

dirkjanfaber, 2023-09-13:
For now we need to wait a few hours to see if the target SOC is indeed being followed by the battery. I'll keep an eye on your site as well.

If you encounter strange things in the mean time, just update this issue.

LecramRed, 2023-09-13:
be aware that I have an battery expansion planned for today, I will double the capacity from 12 to 24 kWh. That can disturb the process somewhat

dirkjanfaber, 2023-09-13:
That should not be a problem if you:
- update to 0.1.2 (which fixes copying the configured value of `b_max` to the context)
- update the battery capacity in the flow

Afterwards you can check in the context data if your flow contains 24 instead of 12 for `b_max` (14 in this example screenshot)

[image]

SimonYoungtree, 2023-09-13:
> Sure. You can import this into Node-RED via the ctrl-i / command-i / via the menu and then pasting it into the text area.
> 
> [image] [image]
> That will give you this extra piece of flow: [image]
> 
> Put it somewhere on the canvas and then redeploy the flow. That should do the trick.

I also did the update and I have added the flow

dirkjanfaber, 2023-09-13:
Looking at a graph of vrm logging, I see that the battery SOC (yellow) is following the target SOC (green) nicely. So I'll close this issue.

[image]

closed 2023-09-13 by dirkjanfaber