victronenergy/dynamic-ess #55: [Bug]: Still showing yesterday prices

LecramRed, 2023-09-13:
### Contact Details

info@redupsell.nl

### VRM / Site iD

345657

### Country / region

Netherlands (nl)

### B max

12

### TB max

3.5

### FB max

4.5

### TG max

15

### FG max

15

### feed-in possible

None

### feed-in possible

None

### What happened?

A new day started (it is now 5 am) and the graph still shows 48 hours and it calculates with yesterday prices. The today prices are shown as tomorrow prices.
I cannot post a screenshot from my tablet

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

LecramRed, 2023-09-13:
After removing the dess context flow, disabling and enabling the flow again, it has been corrected. But still a bug

dirkjanfaber, 2023-09-13:
This should now be fixed (from 0.1.2). If not, please re-open the issue or create a new one.

closed 2023-09-13 by dirkjanfaber