victronenergy/dynamic-ess #58: Question: settings ok?

Mableeker, 2023-09-13:
I've installed version 0.1.2 but the schedule shows no charge or discharge activity this afternoon/ evening, while prices are right to do so. In remote console I see a status unknown, is that ok or should it be changed? [image]
[image]

dirkjanfaber, 2023-09-13:
Status "unknown' is expected. (https://github.com/victronenergy/dynamic-ess#the-console-shows-dynamic-ess-mode-unknown). 

My guess about not charging and discharging is that it probably costs more to do so than it is to keep the batteries around the same level. Note that the aim for Dynamic ESS is not to trade with energy, but to minimize the overall costs. See the [about](https://github.com/victronenergy/dynamic-ess#about) section on the README.

closed 2023-09-13 by dirkjanfaber

Mableeker, 2023-09-13:
Thanks, I missed that trouble shooting section, sorry. And I missed a typo in the sell formula. I was too hasty 🫢

Everything looks good now!

dirkjanfaber, 2023-09-13:
No problem. Glad it got sorted!