victronenergy/dynamic-ess #59: [Bug]: Table shows different trg SOC compared to graph

LecramRed, 2023-09-13:
### Contact Details

info@redupsell.nl

### VRM / Site iD

345657

### Country / region

Netherlands (nl)

### B max

24

### feed-in possible

None

### feed-in possible

None

### What happened?

The table says at 12 and 1AM 13%. The graph shows 33% at 12AM and 34% at 1AM. 

[image]

[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

SimonYoungtree, 2023-09-13:
I have the same problem. Also the order 12AM (This is mid-day) and then 1AM (correct)

dirkjanfaber, 2023-09-13:
Good find. This is a bug in the table. The actual schedule does contain the values that are in the graph. (You can check that by logging in via ssh and starting `dbus-spy` to check the inserted schedule under `com.victronenergy.settings` `Settings/DynamicEss/Schedule/*/Soc`).

This will be fixed with the next update.

LecramRed, 2023-09-13:
The graph is a better scheme than the table. So it is good that the graph is the right one.

LecramRed, 2023-09-13:
It is now 00.30 and unfortunately, dess took the table data to discharge to 13%. So not the graph, but the table was leading. The graphs are still showing 48 hrs, despite a new day. The current day is the data from yesterday. It seems that it relies on yesterday's data, rather than today's data. So something is going wrong here.

LecramRed, 2023-09-13:
The context flow dess still contains 48 hrs data at 1.08 am

LecramRed, 2023-09-13:
I just deleted flow.dess manually, disabled and enabled the tab, and it is 24 hrs of data again. So it still does not refresh the old flow.dess automatically

dirkjanfaber, 2023-09-14:
Thanks for staying up that late. I probably need to that for the next few days too to make sure that I can pinpoint where exactly it goes wrong. On another site I noticed it going wrong from midnight until 6 minutes after midnight. Then the schedule was 24 hours again and everything worked as it should.

dirkjanfaber, 2023-09-14:
This _should_ be fixed by switching to version 0.1.3. Note that you also need to re-import the flow as the code of the _prepare dess variables_ function changed.

closed 2023-09-14 by dirkjanfaber