victronenergy/dynamic-ess #60: [Feature]: Make a difference between DC and AC connected solar with regard to battery costs

SimonYoungtree, 2023-09-14:
### Contact Details

siempoot@gmail.com

### Description

The costs for using the battery are mainly (90%?) caused by using the Multiplus to convert from AC to DC and vice versa. I have both DC and AC connected solar, so I can imagine that there are moments for which it would be smart to dump the AC generated energy straight to the grid, but at the same time charge the battery by using the DC capacity.

### Suggested Solution

Take the difference of charging via DC or AC into consideration (Within the prediction plan) and act accordingly during the day.

SimonYoungtree, 2023-09-15:
I am happy to see that currently the system is charging the battery from DC only. Power from AC is going out to the GRID directly. So this is already implemented?

SimonYoungtree, 2023-10-04:
Is there any plan to consider the "battery" costs to charge from DC connected solar to be 0 (Zero)? I do not think DESS makes optimal use of this.

MichaKersloot, 2023-11-07:
Not sure if I see the same. I only have DC connected PV and at this moment DESS decided to sell to the grid, while the batteries are at 45%. My battery-costs are set to 3 cents, but in the evening the prices are 7 to 8 cents higher than now. As losses in the batteries itself are almost nothing I'm don't understand why DESS decides to sell at this lower price and not charge the batteries and sell it at a higher price in the evening. The losses are virtually the same.
[image]

p.s VRM ID: c0619ab410ae

SimonYoungtree, 2023-11-07:
I have "tricked" my system by setting the ESS discharge limit to zero between 10am and 4pm via Node Red. This way all DC connected solar energy is stored during that time window. The AC connected solar still follows the DESS rules. DESS does not seem to have a problem with this.

MichaKersloot, 2023-11-07:
Sounds like a neat trick.. where did you do this? I seem to be unable to find ESS discharging limit ;-)  Hmm.. I see 'shed charge' instead of 'external control' so you disable DESS and enable the default ESS schedules?

SimonYoungtree, 2023-11-07:
ESS Control Node: Venus settings : Max Inverter Power (W)
can be set to zero or any value your inverters support.

MichaKersloot, 2024-01-08:
As there is since long a sunny day. I noticed the charging power (W) really goes down when the battery gets full. 
Battery Parameters say: CVL: 53.2V, CCL: 40A, DCL: 160A
Solar charger settings: Charge voltage 52.40V, Charge current 100A
Current solar charging: Battery: Voltage 52.40V, Current: 5A

When I change the limit back from 0 to the 4100 I normaly have the Solar charger can produce a lot more power:  Battery: 51.6V, 49A

So this solution is limiting the yield from the system (a lot?)

dirkjanfaber, 2024-01-08:
@MichaKersloot I do notice that you are not running the latest firmware right now. v3.20~17 vs v3.20~35. 
Also noticing that your flow does not contain restriction fields. I suggest updating that too.

MichaKersloot, 2024-01-08:
Hi @dirkjanfaber , thank you for your time. I did try the ~33 version, but had problems with the mqtt p1 connection: https://community.victronenergy.com/questions/244325/venus-os-v32027-available-for-testing.html?childToView=247950#comment-247950 is that solved in ~35 you think?

dirkjanfaber, 2024-01-09:
There where some changes regarding the mqtt broker between \~33 and \~35, so it is worth a try:
- dbus-flashmq, newest master https://github.com/victronenergy/dbus-flashmq
  - Adds support for /Identifier to supplement /DeviceInstance in cases there is more than one of something.
  - Fixes writing arrays as values, including null, which becomes an array.
  - Fix dealing with signals that don't contain the values of paths
  - Fix publishing all known item data on read/getvalue

MichaKersloot, 2024-01-12:
Hi, i've just tried ~37 and there the p1mqqt got connected after a reboot of the esp device, but no data is updated. But mqtt explorer does show data:
[image]

[image]

MichaKersloot, 2024-03-08:
Ok, today I was able to update (took the released 3.22 version). It seems there was a change in https://github.com/freakent/dbus-mqtt-devices needed and the new version of that solved my previously reported problem. I've also updated the node-red stuf and installed the most recent example flow. That one is now running without changes.

Long story short. I'm back for testing.

MichaKersloot, 2024-03-21:
And here my first observations. No clue how the algorithm works, but it surely looks off.
[image]
[image]

For me it now looks like there are no losses in charging from the grid. Or the charging from PV are the same as charging from the grid... Aha ok. Back to disable charging from grid during the day I think.

P.s. Portal ID: c0619ab410ae

MichaKersloot, 2024-03-24:
@SimonYoungtree how are you managing this now? Seems like the update to 3.22 broke the trick to set the inverter power to zero doesn't work anymore?

SimonYoungtree, 2024-03-24:
I am at 3.30 and the "trick" is still working at my site.

MichaKersloot, 2024-04-08:
I've altered the logic a bit to look at PV starts producing instead of time of day and have the current price lower than average.

```
[{"id":"8f445330606365fb","type":"victron-input-solarcharger","z":"5ec7274ba2dd58cb","g":"ea6da7c4489bd0ad","service":"com.victronenergy.solarcharger/273","path":"/Yield/Power","serviceObj":{"service":"com.victronenergy.solarcharger/273","name":"SmartSolar MPPT VE.Can 250/100 rev2"},"pathObj":{"path":"/Yield/Power","type":"float","name":"PV Power (W)"},"name":"","onlyChanges":false,"roundValues":"0","x":300,"y":1060,"wires":[["fd5ec03195b8a8be"]]},{"id":"fd5ec03195b8a8be","type":"rbe","z":"5ec7274ba2dd58cb","g":"ea6da7c4489bd0ad","name":"Filter small changes","func":"deadbandEq","gap":"5%","start":"","inout":"out","septopics":true,"property":"payload","topi":"topic","x":640,"y":1060,"wires":[["483b55b459ce6726"]]},{"id":"483b55b459ce6726","type":"join","z":"5ec7274ba2dd58cb","g":"ea6da7c4489bd0ad","name":"","mode":"custom","build":"object","property":"payload","propertyType":"msg","key":"topic","joiner":"\\n","joinerType":"str","accumulate":true,"timeout":"","count":"2","reduceRight":false,"reduceExp":"","reduceInit":"","reduceInitType":"","reduceFixup":"","x":230,"y":1140,"wires":[["1da48e45d5e34c2b"]]},{"id":"e8d7fd69dc16d657","type":"victron-input-battery","z":"5ec7274ba2dd58cb","g":"ea6da7c4489bd0ad","service":"com.victronenergy.battery/512","path":"/Soc","serviceObj":{"service":"com.victronenergy.battery/512","name":"Pylontech battery"},"pathObj":{"path":"/Soc","type":"float","name":"State of charge (%)"},"name":"","onlyChanges":true,"x":250,"y":1200,"wires":[["483b55b459ce6726"]]},{"id":"1da48e45d5e34c2b","type":"function","z":"5ec7274ba2dd58cb","g":"ea6da7c4489bd0ad","name":"Inverter settings","func":"let dess = flow.get('dess');\nlet date = new Date();\n\nvar PVPower = msg.payload[\"SmartSolar MPPT VE.Can 250/100 rev2 - PV Power (W)\"];\nvar BatSOC =  msg.payload[\"Pylontech battery - State of charge (%)\"]; \nvar lastValidUpdate = new Date(flow.get('lastValidUpdate')).toLocaleString();\nvar hour = date.getHours();\n\n\nfunction calculateAverage(array) {\n    var total = 0;\n    var count = 0;\n\n    array.forEach(function(item, index) {\n        total += item;\n        count++;\n    });\n\n    return total / count;\n}\n\nvar prices = Object.values(dess.output.p_s);\n\nvar priceavg = calculateAverage(prices);\nvar pricenow = prices.slice(hour, hour+1)[0];\nvar InverterPower = 4100;\n\n//node.warn ('price avg: ' + priceavg + '| price now: ' + pricenow )\n\nif (BatSOC < 93 && pricenow < priceavg) {\n    if (PVPower > 100) {\n        InverterPower = 0;\n    } else {\n        InverterPower = 4100;\n    }\n}\n\nnode.status({ fill: 'green', shape: 'dot', text: InverterPower + ' Pnow:' + parseFloat(pricenow).toFixed(2) + ' Pavg:' + parseFloat(priceavg).toFixed(2)});\n\nreturn { payload: InverterPower};","outputs":1,"timeout":0,"noerr":0,"initialize":"","finalize":"","libs":[],"x":400,"y":1140,"wires":[["39bc68fea0f0a45d"]]},{"id":"39bc68fea0f0a45d","type":"rbe","z":"5ec7274ba2dd58cb","g":"ea6da7c4489bd0ad","name":"Pass changes only","func":"rbe","gap":"","start":"","inout":"out","septopics":true,"property":"payload","topi":"topic","x":630,"y":1140,"wires":[["b40ffb46a8af08bc"]]},{"id":"b40ffb46a8af08bc","type":"victron-output-ess","z":"5ec7274ba2dd58cb","g":"ea6da7c4489bd0ad","service":"com.victronenergy.settings","path":"/Settings/CGwacs/MaxDischargePower","serviceObj":{"service":"com.victronenergy.settings","name":"Venus settings"},"pathObj":{"path":"/Settings/CGwacs/MaxDischargePower","type":"integer","name":"Max inverter power (W)","writable":true},"initial":"","name":"","onlyChanges":false,"x":1000,"y":1140,"wires":[]}]
```