victronenergy/dynamic-ess #61: [Bug?]: Why charging with the handbrakes on?

SimonYoungtree, 2023-09-14:
### Contact Details

siempoot@gmail.com

### VRM / Site iD

c0619ab31cc4

### Country / region

Netherlands (nl)

### B max

40

### TB max

6

### FB max

6

### TG max

16

### FG max

16

### Battery costs

0.027

### Buy price

(p+0.13)*1.21+0.02

### Sell price

(p+0.13)*1.21-0.02

### feed-in possible

yes

### feed-in possible

None

### What happened?

A bug happened, (but it may also be a feature)! During midday, when the prices are lowest the system is charging, but at a limited speed. Is this limited speed a feature or a bug? If it is a feature, then what is the thought behind? (Perhaps to avoid losses in the Multiplus?)
[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-14:
Have you checked [here](https://github.com/victronenergy/dynamic-ess/tree/main#the-battery-discharges-too-slow-to-keep-up-with-the-target-soc)?

The system needs to know your battery capacity in order to switch up gears.

closed 2023-09-14 by dirkjanfaber

SimonYoungtree, 2023-09-14:
I did not see the 0.1.3 version yet. I was still on 0.1.2. Still charging is going up and down. Sometimes reaching max capacity (10A AC per Multiplus configured), but most of he times staying at 2/3rd. This may be by design. I do not know.