victronenergy/dynamic-ess #63: [Bug]: Can not install 1.1.3

Marous23, 2023-09-14:
### Country / region

Czech Republic (cz)

### feed-in possible

None

### feed-in possible

None

### What happened?

Install fails. Here I found the complete log:
root@einstein:~# cat /data/home/nodered/.npm/_logs/2023-09-14T13_26_30_824Z-debug-0.log 
0 verbose cli /usr/bin/node /usr/bin/npm
1 info using npm@9.5.1
2 info using node@v18.16.0
3 timing npm:load:whichnode Completed in 46ms
4 timing config:load:defaults Completed in 24ms
5 timing config:load:file:/usr/lib/node_modules/npm/npmrc Completed in 15ms
6 timing config:load:builtin Completed in 17ms
7 warn config production Use `--omit=dev` instead.
8 timing config:load:cli Completed in 37ms
9 timing config:load:env Completed in 2ms
10 timing config:load:file:/data/home/nodered/.node-red/.npmrc Completed in 1ms
11 timing config:load:project Completed in 12ms
12 timing config:load:file:/data/home/nodered/.npmrc Completed in 1ms
13 timing config:load:user Completed in 2ms
14 timing config:load:file:/usr/etc/npmrc Completed in 2ms
15 timing config:load:global Completed in 4ms
16 timing config:load:setEnvs Completed in 13ms
17 timing config:load Completed in 120ms
18 timing npm:load:configload Completed in 124ms
19 timing npm:load:mkdirpcache Completed in 3ms
20 timing npm:load:mkdirplogs Completed in 2ms
21 verbose title npm install victron-dynamic-ess@0.1.3
22 verbose argv "install" "--no-audit" "--no-update-notifier" "--no-fund" "--save" "--save-prefix" "~" "--production" "--engine-strict" "victron-dynamic-ess@0.1.3"
23 timing npm:load:setTitle Completed in 18ms
24 timing config:load:flatten Completed in 45ms
25 timing npm:load:display Completed in 75ms
26 verbose logfile logs-max:10 dir:/data/home/nodered/.npm/_logs/2023-09-14T13_26_30_824Z-
27 verbose logfile /data/home/nodered/.npm/_logs/2023-09-14T13_26_30_824Z-debug-0.log
28 timing npm:load:logFile Completed in 69ms
29 timing npm:load:timers Completed in 2ms
30 timing npm:load:configScope Completed in 1ms
31 timing npm:load Completed in 350ms
32 timing arborist:ctor Completed in 11ms
33 silly logfile start cleaning logs, removing 2 files
34 silly logfile done cleaning log files
35 timing idealTree:init Completed in 122ms
36 timing idealTree:userRequests Completed in 60ms
37 silly idealTree buildDeps
38 silly fetch manifest victron-dynamic-ess@0.1.3
39 timing arborist:ctor Completed in 2ms

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-14:
This logging does not provide enough information for me to check on what might be going wrong. How did you try to install? From the palette manager?

dirkjanfaber, 2023-09-14:
This can by kinda tricky to get going again. 
Can you clean the npm cache and try again? That should be do-able from an ssh session in `/data/home/nodered/.node-red/` and running `npm cache clean --force`.

Marous23, 2023-09-14:
Finally got it to working state.
done the:
**npm cache clean --force**
**rm -Rf /data/home/nodered/.node-red/**
restarted service
Then I was able to install 0.1.3, import, config, etc.

Thanks for the tip.

closed 2023-09-14 by Marous23