victronenergy/dynamic-ess #65: [Bug]: Max power to Grid

fofo444, 2023-09-15:
### Contact Details

fofo444@gmail.com

### VRM portal ID

b827eb779d5c

### Country / region

Slovakia (sk)

### B max

14

### TB max

5

### FB max

5

### TG max

2.5

### FG max

2.5

### Battery costs

0.04

### Buy price

0.254

### Sell price

p

### feed-in possible

yes

### feed-in possible

yes

### What happened?

Setting max power baterry to grid 2500. Maximum feed-in 2800w. Real delivery to grid 3136w. 

[image]
[image]
[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-15:
Thanks for reporting and the clear report and screenshots. We'll take a look at it next week.

dirkjanfaber, 2023-09-19:
Duplicate of issue #69 .

closed 2023-09-19 by dirkjanfaber

fofo444, 2023-09-20:
It's not a duplication. he problem persists. This problem is when discharging the battery to the grid. I can supply a maximum of 3kW. I have set a limit of 2.5kW. And the system supplies 3.1 kW.

fofo444, 2023-09-20:
Tested on candidate version. Into the grid goes 2.8kW. Dess set to 2.5. In the graph is 2.5. Is it okay?