victronenergy/dynamic-ess #66: [Bug]: Frequent periods of full power feed-in although feed-in is not allowed

beliffm, 2023-09-16:
### Contact Details

beliffm@gmail.com

### VRM portal ID

c0619ab0327f

### Country / region

Germany (de)

### B max

28

### TB max

6.5

### FB max

4.5

### TG max

0

### FG max

2.5

### Battery costs

0.05

### Buy price

(p*1.03+0.15)*1.19

### Sell price

default (greyed out)

### feed-in possible

no

### feed-in possible

no

### Version

0.13

### What happened?

After start of the flow the Ess feeds-in full throttle for some seconds an then stops and recharges the batterys. This repeats frequently but not regularly every 5 to 15 minutes.
As can be seen in the log excerpt both values &feed_in_possible=true&feed_in_control_on=true although unselected during data input.
After initially being green (OK) the Dynamic-ESS flow shows red (Invalid VRM Token) a lot but not all the time. 
 
[image]

### Relevant log output

```shell
@400000006505caa633702cbc 16 Sep 17:32:44 - [warn] [victron-dynamic-ess:Wick81-Dyn-ESS] [object Object]
@400000006505caa633cd8b14 curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa633d9857c curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa633df2eb4 curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa633e4bc94 curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa633e95844 curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa633edd4b4 curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa633f287d4 curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa633f6d94c curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa633fb26dc curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa633fff16c curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa63403d19c curl -X GET -H "Accept:application/json" -H "Content-Type:undefined" -H "X-Authorization:Token 845288a9671082140368e2d98cbca83b740f470686079560ae205a5cba5e5ae7" -H "User-Agent:dynamic-ess/0.1.3" "https://vrm-dynamic-ess-api.victronenergy.com?vrm_id=c0619ab0327f&b_max=28&tb_max=6.5&fb_max=4.5&tg_max=0&fg_max=2.5&b_cost=0.04&buy_price_formula=(p*1.03+0.15)*1.19&sell_price_formula=(p-0.03+0.13)*1.21&feed_in_possible=true&feed_in_control_on=true&country=DE"
@400000006505caa704f9df54 16 Sep 17:32:45 - [warn] [victron-dynamic-ess:Wick81-Dyn-ESS] AxiosError: Request failed with status code 401
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-17:
A http code 401 means "unauthorized". Do you have the VRM access token correct?

See https://github.com/victronenergy/dynamic-ess#create-an-access-token

beliffm, 2023-09-17:
Access token is correct but the indicator in the flow is all over the place. Sometimes it is green (OK), sometimes it is yellow (Trying to update too quickly...) and most of the time it is red (Invalid VRM token) Maybe due to trying to update too frequently the updates get blocked.

dirkjanfaber, 2023-09-19:
Updating too quickly indeed blocks on the API side. Also there is no need to update more frequently (calculations are being updated every 15 minutes IIRC).

beliffm, 2023-09-19:
Mind You, the (too) high update rate was not induced by me, i installed the dynamic-ess flow according to the instructions provided.

beliffm, 2023-09-19:
Since i never used the previous versions, there should only be the latest one.

dirkjanfaber, 2023-09-19:
Ok, thanks for clarifying. Let me re-open the issue and I'll ask a colleague to take a look at the API logging for your site.

beliffm, 2023-09-19:
While you are at it, it seems to me that only the first input of the configuration is considered. Any subsequent changes are discarded. And the feed-in flags are always true, regardless of the input. Not sure whether that is another issue though.

dirkjanfaber, 2023-09-19:
That is the issue that I've been looking into for the last hour. About to release an update for that.

dirkjanfaber, 2023-09-19:
Released the update. Should be available via the palette manager shortly (it sometimes takes a few hours until the Node-RED cache has been updated).

Can you update and check if that fixes your problem?

beliffm, 2023-09-19:
Will do.

beliffm, 2023-09-19:
Looks good! Changes are deployed, feed-in flags are set to the values i put in.  And even the update rate seem to be in order now, the indicator is solid green, only interrupted by a short yellow (Retrieving setpoint) every now and then. I have injected the enable command and keep watching.

beliffm, 2023-09-19:
I changed the selling price to test before i sent the comment above and it seemed to be OK by then. What i found now is that all three test values are sent round-robin by the flow now.  Looks like they all get stored. I pasted the debug output below, where you can see all three values without my interference.

19.9.2023, 22:34:49[node: Wick-Dyn-Ess](https://192.168.8.252:1881/#)msg : Object
object
url: "https://vrm-dynamic-ess-api.victronenergy.com"
options: object
vrm_id: "c0619ab0327f"
b_max: "28"
tb_max: "6.5"
fb_max: "4.5"
tg_max: "0"
fg_max: "2.5"
b_cost: "0.04"
buy_price_formula: "(p*1.03+0.15)*1.19"
sell_price_formula: "0.5"
feed_in_possible: "false"
feed_in_control_on: "false"
country: "DE"
headers: object
19.9.2023, 22:39:22[node: Wick-Dyn-Ess](https://192.168.8.252:1881/#)msg : Object
object
url: "https://vrm-dynamic-ess-api.victronenergy.com"
options: object
vrm_id: "c0619ab0327f"
b_max: "28"
tb_max: "6.5"
fb_max: "4.5"
tg_max: "0"
fg_max: "2.5"
b_cost: "0.04"
buy_price_formula: "(p*1.03+0.15)*1.19"
sell_price_formula: "(p-0.03+0.13)*1.21"
feed_in_possible: "false"
feed_in_control_on: "false"
country: "DE"
headers: object
19.9.2023, 22:40:38[node: Wick-Dyn-Ess](https://192.168.8.252:1881/#)msg : Object
object
url: "https://vrm-dynamic-ess-api.victronenergy.com"
options: object
vrm_id: "c0619ab0327f"
b_max: "28"
tb_max: "6.5"
fb_max: "4.5"
tg_max: "0"
fg_max: "2.5"
b_cost: "0.04"
buy_price_formula: "(p*1.03+0.15)*1.19"
sell_price_formula: "0.2874"
feed_in_possible: "false"
feed_in_control_on: "false"
country: "DE"
headers: object

dirkjanfaber, 2023-09-19:
This is no good; How do you have your deploy set? To all nodes or to the modified nodes?

[image]

beliffm, 2023-09-19:
Full.... could it be that i should have deployed only the changed ones? And by deploying full i duplicated the whole flow?

dirkjanfaber, 2023-09-20:
No, I was looking at the wrong thing. I should have focused on the timers more, where the actual bug was. It handled cleaning up the used timers wrong. I've just released `0.1.5`, so please update once more.

dirkjanfaber, 2023-09-20:
That is very good to hear. Closing this issue. If something new pops up, please open a new one.

closed 2023-09-20 by dirkjanfaber