victronenergy/dynamic-ess #67: [Bug]: dynamic ess starts to discharge my battery to grid

disaster123, 2023-09-17:
### Contact Details

through github

### VRM portal ID

e45f019ec891

### Country / region

Germany (de)

### B max

10

### TB max

4

### FB max

4

### TG max

4

### FG max

4

### Battery costs

0.01

### Buy price

(p+0.1574)

### Sell price

0.07

### feed-in possible

yes

### feed-in possible

yes

### Version

0.1.3

### What happened?

When i enable Dynamic ESS - it starts to immeaditly discharge the battery to grid - even the battery is not full and i would need the power for night...

Or is it due to the fact that i see no forecast in vrm? I added grid and pvinverter just 3 days ago.

### Relevant log output

```shell
no idea how to get the log. I did not find any information how to get a log. Never used Node Red before... i'm more in python or C ;-)
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-19:
Please wait until there is one week of forecast data and try again. At the moment the models require a week of data before to function properly. 
If the problem persists, please re-open the issue (or file a new one).

closed 2023-09-19 by dirkjanfaber

disaster123, 2023-09-19:
Ok thanks may be update the documentation which says 48hours.