victronenergy/dynamic-ess #68: VRM token not possibble

mike05-1976, 2023-09-17:
### Contact Details

mike vugts

### VRM portal ID

c0619ab3a747

### Country / region

Netherlands (nl)

### B max

15

### TB max

4

### FB max

14.4

### TG max

7

### FG max

17.25

### Battery costs

0.02

### feed-in possible

yes

### feed-in possible

yes

### Version

lates

### What happened?

dear lady/gentleman
I had a vrm token but I lost it, now I want to log in again and created a new token and the vitronengy does not want to do this.
what should I do to resolve this?

Sincerely
Mike

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-18:
Your report is a bit unclear, so some clarification is needed in order to help you further.

A new access token can be created via https://vrm.victronenergy.com/access-tokens

If that does not work, can you provide some more information on what is going wrong? Error messages / screenshots will help.

closed 2023-09-19 by dirkjanfaber

mike05-1976, 2023-09-21:
[image]

mike05-1976, 2023-09-21:
[image]