victronenergy/dynamic-ess #69: [Bug]: Battery charging with full solar power

fofo444, 2023-09-17:
### Contact Details

fofo444@gmail.com

### VRM portal ID

b827eb779d5c

### Country / region

Slovakia (sk)

### B max

14

### TB max

5

### FB max

5

### TG max

2.5

### FG max

2.5

### Battery costs

0.04

### Buy price

0.254

### Sell price

p

### feed-in possible

yes

### feed-in possible

yes

### What happened?

When battery charging is planned, it is with the full power of the solar panels and the appliances are powered from the grid. 
(Battery charging) = (Pv charger)  and (Grid) = (AC Loads)
It is rightly  (Battery charging) = (Pv charger) - (AC Loads) and (Grid) = 0

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-19:
Thanks for the report. We are looking into it. Do you have feed-in of excess DC coupled PV enabled on your system?

fofo444, 2023-09-19:
[image]

dirkjanfaber, 2023-09-19:
We've identified an issue in the underlying code, which will be resolved with the next candidate release.

closed 2023-09-19 by dirkjanfaber