victronenergy/dynamic-ess #70: [Feature]: Planned battery SOC at evening.

fofo444, 2023-09-17:
### Contact Details

fofo444@gmail.com

### Description

I would like to set the planned SOC battery will have in the evening. 
My safe SOC value is 60%.
The system has no priority to achieve this in the evening. 
According to observation, the battery starts to charge during the lowest price. As long as it can charge, the SOC will be in the evening. 

### Suggested Solution

I suggest to start charging the battery in the morning. Priority to have SOC more than set in the evening. In the picture I marked in red. 

[image]

DennisBosmans, 2023-09-20:
This disables the grid when the battery below 60% SOC and turns it back on when battery is lower then 5% SOC : 

[image]

Function : 

if(msg.payload >60 || msg.payload < 5){
       msg.payload = (3);
   }
       
else{
      msg.payload = (2);
   }
return msg;

fofo444, 2023-09-20:
Thank you, very simple and inspiring. The problem is only in overload. Then the whole system crashes.  I have a Multiplus II 5kW and sometimes it happens that I overload it.  Any solution that would also work with the grid connected?

fofo444, 2023-09-20:
Only my usage above 5kW.  It is a overload of the inverter.

DennisBosmans, 2023-09-21:
Add this if it's a one fase system : 
[image]

This function : 

if(msg.payload >4900){
    global.set("state", (3));
}
else{
    global.set("state",(2));
}

Function from previous message : 

var state = global.get("state")
if(msg.payload >60 || msg.payload < 5 || state == 3){
msg.payload = (3);
}

else{
msg.payload = (2);
}
return msg;