victronenergy/dynamic-ess #72: Help us make a predefined lists of formulas

mpvader, 2023-09-19:
Hi all,

To make it easier to set a Dynamic ESS system up, we'd like to have a predefined list of formulas. Can you help us by sharing your formula? These are the fields we need to know:

- Country
- Bidding zone (some countries have that, like Sweden and Italy).
- Energy provider
- Buy price formula & Sell price formula

For example:

- The Netherlands
- Provider: ANWB
- No bidding zone
- Buy price formula: `(p+0.03+0.13)*1.21`
- Sell price formula: `(p-0.03+0.13)*1.21`

halma1, 2023-09-19:
The Netherlands
    Provider: Zonneplan
    No bidding zone
    Buy price formula: (p+0.02+0.13)*1.21
    Sell price formula: (p-0.02+0.13)*1.21

khostri, 2023-09-19:
I will add additional info, as its not so simple here in Czech Republic :)
There are 3 distribution areas, different price composition depending on chosen rate (same price 24/7, 8 hour lower price or 20 hours lower price). 
Also final price is combined from market price and from distribution price. Distribution price has some price part per kWh, some prices per month. In the data below I will post harmonized prices I have created, but also add picture of table to better understand. Monthly prices are transformed to kWh prices based on monthly average consumption and monthly average sell amount of kWh.

Country: Czech Republic
Provider: ENTRI
Distribution Area: CEZ 
No bidding zone
Buy price formula: p+0.005474217+0.001126554+0.009732008+0.143210846+0.00136447+0.013301485+0.011955049
Sell price formula: p+0.002222509+0.019925082

[image]

LecramRed, 2023-09-19:
The Netherlands
Provider: Tibber
No bidding zone
Buy price formula: (p+0.13)*1.21+0,025
Sell price formula: (p+0.13)*1.21+0.015

halma1, 2023-09-20:
> The Netherlands Provider: Tibber No bidding zone Buy price formula: (p+0.13)*1.21+0,025 Sell price formula: (p+0.13)*1.21+0.015

Is there no btw on the 0,025 ?

(p+0.025+0.13)*1.21

MarcoFace, 2023-09-20:
The Netherlands
Provider: ANWB 
No bidding zone
Buy price formula: (p+.0175+.12175)*1.21
Sell price formula: ((p+.0175+.12175)*1.21)/1.20

N.B. The sell price is discounted by 20% because a roundtrip of charging and discharging the batteries loses the amount of Kwh you can sell back to the grid by 20%. As the algorithm does not seem to take into account these considerable losses I off set the loss in the price to ensure the system does not arbitrage at an actual loss when the spread is low.
Ideally there is a separate parameter (or dynamically derived parameter from actual site data) for the round trip loss. The battery costs parameter is not suitable as it is a PRICE parameter while the system loses Kwh (QUANTITY) and it is intended to factor in battery depreciation rather than inefficiency.

SimonYoungtree, 2023-09-20:
> ```
> The Netherlands
> Provider: Zonneplan
> No bidding zone
> Buy price formula: (p+0.02+0.13)*1.21
> Sell price formula: (p-0.02+0.13)*1.21
> ```

For Zonneplan: the formula should actuallu be (p+0.13)*1.21+(or -) 0.02, since the fee is already including VAT

SimonYoungtree, 2023-09-20:
> The Netherlands Provider: ANWB No bidding zone Buy price formula: (p+.0175+.12175)*1.21 Sell price formula: ((p+.0175+.12175)*1.21)/1.20
>
> N.B. The sell price is discounted by 20% because a roundtrip [...]

The cost of inefficiency can be included in the costs of battery usage: single trip losses result in ~2.7 (EURO) cent if the average price per kwh is 27 cent. But Victron could calculate this themselves: the majority of the losses are from the Multiplus and I assume Victron knows best what the losses are at a given though put.

andressade, 2023-09-20:
Country: Estonia

rsm = provider sell margin (depends on the contract)
rbm = provider buy margin (depends on the contract)
gf = grid fee. Day and night prices are different. Day 7:00 a.m. to 10:00 p.m., night 10:00 p.m. to 7:00 a.m. and weekends and may be ( depends on the contract) different on day peak times ( weekdays 9:00 AM to 12:00 PM and 4:00 PM to 8:00 PM) and weekend peaks times (In effect from November to March, on Saturdays, Sundays, and national holidays, from 4:00 PM to 8:00 PM.)

Buy price formula: ( p + rsm + gf )*1.20
Sell price formula:  p - rbm

DennisBosmans, 2023-09-20:
Country : Belgium 
Energy provider : Engie
Contract Type : Engie Dynamic

Buy price : (p+0.02+0.13)*1.06
Sell price  : p

MarcoFace, 2023-09-20:
> The cost of inefficiency can be included in the costs of battery usage: single trip losses result in ~2.7 (EURO) cent if the average price per kwh is 27 cent. But Victron could calculate this [...]

The dynamic sell price varies day by day to a large extent. By nature it is the most extreme pricing of the day otherwise the algorithm would not be selling at that point. Hence it would be much more accurate and elegant to discount the round trip inefficiency on the quantity side rather than the price side. By sampling a few cycles of data the algorithm can deduce a site's efficiency at a round trip and use this to calculate its schedule. This way we do not have to abuse the sell price formula to correct for it or use the battery cost parameter with great inaccuracy.

LecramRed, 2023-09-20:
> > The Netherlands Provider: Tibber No bidding zone Buy price formula: (p+0.13)*1.21+0,025 Sell price formula: (p+0.13)*1.21+0.015
> 
> Is there no btw on the 0,025 ?
> 
> (p+0.025+0.13)*1.21

No, the 0.025 is VAT included. In fact it is 0.02 for selling and buying but I adjusted it a but because of the charge/discharge losses.

KajLehtinen, 2023-09-20:
Sweden:
Grid owner MSE, hourly prices through Tibber.

Buy price formula: (p+0.027692+0.041281)*1.25
Sell price formula: (p-0.00+0.05139)*1.25

The Buy 0.027 is the grid owners charge which varies with all grid owners in Sweden, the 0.041 is the tax.
The Sell price includes what the grid owner pays (~0.001) and the rest is the tax deduction that we get here in Sweden.

Since our 'fees' are in local SEK the fixed amounts varies depending on the exchange rate compared to P which is calculated in Euro. Perhaps insert a option to select currency and translate that behind the scene?

koleo9am, 2023-10-12:
Is there a possibility to have different distribution fees in peek and off-peek periods?

ajnadox, 2023-10-13:
@mpvader  as a respons to @KajLehtinen question the Free for public use API could be used (SEB is a Swedish bank) https://developer.sebgroup.com/products/fxrates

Yes, it is possible to select a currency and translate that behind the scene. This would allow you to use the same Buy And Sell formulas for Sweden, even though the fixed amounts are in SEK.

To do this, we would need to add a new parameter to the Buy And Sell formulas, which would be the currency. The formula would then use this parameter to convert the fixed amounts to the correct currency before calculating the final price.

For example, if the currency is set to SEK, the Buy And Sell formulas would be:
`buy_price_formula = "(p+0.027692+0.041281)*1.25"
sell_price_formula = "(p-0.00+0.05139)*1.25"
`

However, if the currency is set to EUR, the Buy And Sell formulas would be:
`buy_price_formula = "(p+0.027692*exchange_rate+0.041281*exchange_rate)*1.25"
sell_price_formula = "(p-0.00*exchange_rate+0.05139*exchange_rate)*1.25"
`

Where exchange_rate is the current exchange rate between SEK and EUR.
This would allow you to use the same Buy And Sell formulas for Sweden, regardless of the currency that you are using

bjoernh, 2023-10-16:
Germany Tibber (LEW distribution grid) hourly pricing

buy price = (p+0.1598)*1.19

but the 0.1598 part is locally different, because the prices of the distribution grid differs and are included in this service fee

tuxedo0801, 2023-10-17:
* Germany 
* provider: Tibber (NetzeBW distribution grid) hourly pricing

[image]

* buy price = (p+0.1613)

as @bjoernh mentioned: the 0.1613 part may change depending in region / distribution grid.

* Sell price in Germany depends on your PV installation, size and installation-date. For must users, this will be a fixed value per kWh.

MichaKersloot, 2023-10-24:
The Netherlands
Provider: ANWB
No bidding zone
Buy price formula: (p+inkoopkosten+energiebelasting)*1.21
Buy price formula: (p+0.02118+0.15246)*1.21
Sell price formula: (p+0.02118+0.15246)*1.21

https://www.anwb.nl/huis/energie/anwb-energie/veelgestelde-vragen at "Welke terugleververgoeding ontvang ik?" it states you get both the 'inkoopvergoeding' and 'belastingen' back. although only for the electricity you've used, but that's not easy to catch in a formula.

franklouwers, 2023-10-24:
@DennisBosmans : how did you get to those?

> Country : Belgium Energy provider : Engie Contract Type : Engie Dynamic
> 
> Buy price : (p+0.02+0.13)*1.06
> Sell price : p

I was looking at https://www.engie.be/api/engie/be/ms_1/pricing/v1/public/pricesAndConditionsPDF?document=E_DYNAMIC_R_GREY_C_I_12_V_N&monthOffset=0&segment=R&language=N and fail to see how to build the formula.

From what I see: 

buy: (0.204 + p) * 1.06
sell: p * 1.06 (although you're probably right, there is no vat on sell).

But then how do with all the taxes:
- WKK (2.241 c/WkW) (applies only on buy)
- net fee (depending on the region between 3.6 and 4.98 c/kWh)
- taxes of 0.20417 + 5.0.3288 c/kWh

jirimatta, 2023-11-05:
Czech republic

buy : (p+0.012+0.008)*1.21 (dynamic price + (distribution + seler price))*VAT
sell : fixed price

please add the option of two prices depending on the time, we have two prices for distribution during the day

if I give the price for the battery 0.01 dess will not charge the battery even if the market price is 0€

mman, 2023-11-06:
@jirimatta What energy provider and tariff is that please? We have many in CZ, so to keep that clear.

jirimatta, 2023-11-06:
> @jirimatta What energy provider and tariff is that please? We have many in CZ, so to keep that clear.

Edited in previous post

Linusten, 2023-11-16:
Germany
provider: Tibber (ngn netzgesellschaft niederrhein mbh) hourly pricing
[image]

buy price according to screenshot = (p*1.19) + 0.1501

Sell price in Germany depends on your PV installation, size and installation-date. For must users, this will be a fixed value per kWh.
For my installation it is a sell price of 0.082€

Govraemd, 2023-11-19:
> @DennisBosmans : how did you get to those?
>
>
> I was looking at [...]

how did you fix the taxes? Added them to the formula (0,204 + p + taxes)?

franklouwers, 2023-11-20:
@Govraemd I haven't yet

ThomasDK-cybertruck, 2023-11-26:
> Hi all,
>
> To make it easier to set a Dynamic ESS system up, we'd like to have a predefined list of formulas. Can you help us by sharing your formula? These are the fields we need to know:
>
> * Country [...]

Denmark here

I uses elforbundet and they use vindstoed. 
There are no extra fees.
The prices are:
? I will return 
?  (Energinet)
VAT 1.25 (25%)

MrYoshii, 2023-12-01:
I'm not sure if my formula is correct.

Energy provider country | Austria
Energy provider bidding zone | Austria
Energy provider name | Awattar
Price calculation | (p+(p*0.03)+1.5)*1.20

Stündliche Preise EPEX Spot ® AT
+ | Stündliche Preise EPEX Spot ® AT | * 0.03
+ 1,500 Cent/kWh
+ 20% MwSt.

DennisBosmans, 2023-12-01:
If you count max grid costs in Belgium you will end approx at 0,13€/kWh till the end of december. Next year grid costs will rise gradually to 0,2425€/kWh by the end of 2024. Dennis BosmansSent  from my iPhone

AndreasStEn, 2023-12-15:
Hi guys,

für Austria:

**all costs without the state subsidies valid in 2023** 

**1. Without gridcosts & fees, incl tax:**
Buy
Energy provider country | Austria
Energy provider bidding zone | 
Energy provider name | smartEnergy https://www.smartenergy.at/smartcontrol
Price calculation | p*1.2+0.0144

**2. With gridcosts &  fees, incl tax:** 
Buy
Energy provider country | Austria
Energy provider bidding zone | 
Energy provider name | smartEnergy https://www.smartenergy.at/smartcontrol
Price calculation | p*1.2+0.0144+(0.0703+0.00398+0.015)*1.2

StaKai-CZE, 2024-01-04:
Czech republic
Distribution Area : CEZ
Distribution tariff : D45d

When you are on SPOT dynamic prices of electricity you should add distribution fee which is time based. E.g. for this distribution tariff it is 20 hours of lower fee and 4 hours of higher fee. In my case it is
00:00-08:00 low distribution fee
08:00-09:00 high distribution fee
09:00-12:00 low distribution fee
12:00-13:00 high distribution fee
13:00-15:00 low distribution fee
15:00-16:00 high distribution fee
16:00-19:00 low distribution fee
19:00-20:00 high distribution fee
20:00-00:00 low distribution fee

but this schedule can vary based on ripple control system (HDO in czech) signal.

Formula: not available because this is not possible to set up now. (mix of dynamic prices with time schedule for fees)

vanwalsum, 2024-03-01:
What are the actual formula's for buying and selling with ANWB (Netherlands). I have a doubt if i used the right formula.

[image]
[image]
[image]

dirkjanfaber, 2024-03-01:
> What are the actual formula's for buying and selling with ANWB (Netherlands). I have a doubt if i used the right formula.

I think you are missing some brackets. I'd say `(p+0.0484+0.13)*1.21` and `(p-0.0484+0.13)*1.21` are currently the formula's for ANWB. But I should add that there is no consensus on the exact numbers. Could be that different users get different contracts, but some do offset the p with `0.02` or `0.03` instead of your `0.0484`). The truth is probably in the "kleine lettertjes" of the contract.

kristofdupon, 2024-03-23:
ECOPOWER BELGIUM
BUY PRICES 
Energy provider country | Belgium
Energy provider name | Ecopower
Price calculation : (1.02*p+0.004)*1.21

SELL PRICES
Energy provider country	Belgium
Energy provider name	Ecopower
Price calculation	(0.85*p-0.004)*1.21

Delusional112, 2024-03-23:
I get an error with the predefined sell formula for Tibber. 

[image]

Kositch, 2024-03-23:
Any possibility to make "p" not negative? My provider does not use negative market price, so I would need to change negative price to 0?

Robbot50, 2024-03-24:
**Energy supplier EnergyZero (ditto ANWB) for NL**

Prices per kWh for 1 hour

2024 Buy costs = 0.0484   Please note this is already incl. VAT

2024 Energy tax = 0.1316   Please note this is already incl. VAT

2024 Epex hourly price = P excl. VAT

2024 VAT = 1.21

For the Dutch Energy tax see:
Https://www.belastingdienst.nl/wps/wcm/connect/bldcontentnl/belastingdienst/zakkelijk/oterige_taxen/belastingen_op_milieugrondslag/tarifes_milieubelastingen/tabellen_tarifes_milieubelastingen

Buy price : (P x 1.21) + 0.0484 + 0.1316

Sell price : (P x 1.21)

Sell price without “saldering” (refund purchasing costs and energy tax)

Note:
Sell price (P x 1.21) - (0.0484 + 0.1316) can also be used but with the following side note!
Selling price incl. discount, (refund of purchasing costs and energy tax per kWh), based on “Salderingsregeling”) can be applied if one is sure that the total number of kWh feed-in on an annual basis does not exceed the number of kWh of feed-out (delevery).
If there is more feed-in then for the difference (= surplus) is the sell price is: P + VAT

Unfortunately, this condition cannot be included in the current DESS formula!

In my opinion, the ANWB formulas mentioned above and the current ANWB formula in the DESS setting are not correct, due to double VAT calculation.

I look forward to all reactions.

Ales-Svoboda, 2024-03-27:
> I will add additional info, as its not so simple here in Czech Republic :) There are 3 distribution areas, different price composition depending on chosen rate (same price 24/7, 8 hour lower price [...]

Is the sales formula correct? I think the formula should be as follows:
`sellPrice = p-0.002222509-0.019925082`
Because the seller also pays the distribution on the sale...

thomasraabo, 2024-03-29:
Do i put in the calculation in local currency?

Delusional112, 2024-04-01:
Hi,

Hereby the Tibber formula with the latest price change as of today. 

Ps. The pre defined Tibber formulas in VRM don’t work. I get an error at the sell formula. Maybe it can be fixed with these latest.

[image]
[image]

Robbot50, 2024-04-02:
> Hi,
> 
> Hereby the Tibber formula with the latest price change as of today.
> 
> Ps. The pre defined Tibber formulas in VRM don’t work. I get an error at the sell formula. Maybe it can be fixed with these latest.
> 
> ![IMG_0399](https://private-user-images.githubusercontent.com/79289457/318390727-5dc1f2f2-987d-41b6-b4d9-05c2826d4122.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIwNjgxNTAsIm5iZiI6MTcxMjA2Nzg1MCwicGF0aCI6Ii83OTI4OTQ1Ny8zMTgzOTA3MjctNWRjMWYyZjItOTg3ZC00MWI2LWI0ZDktMDVjMjgyNmQ0MTIyLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwMlQxNDI0MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YjQ5ZGQ2NGY3MDlmOGY3Yjk0MzhjN2IzY2ZjNzcyOTY1MTUzYWMyNzA2NzQwNGYzZWQ2YTY3NGRmY2Q1OWU4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.r1wOiJvKVk534kcdbKvGv3hZIpfE-FJTJaXOhfHVm-A) ![IMG_0400](https://private-user-images.githubusercontent.com/79289457/318390732-7dda8da7-b06c-41bf-a8b6-cd6cf3ba56ab.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIwNjgxNTAsIm5iZiI6MTcxMjA2Nzg1MCwicGF0aCI6Ii83OTI4OTQ1Ny8zMTgzOTA3MzItN2RkYThkYTctYjA2Yy00MWJmLWE4YjYtY2Q2Y2YzYmE1NmFiLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwMlQxNDI0MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMzMzYzkwYzE5MDA5OGQ5NzY0YTZkZGU0MGY0MjQzYzI3MWRlMjU4YWRlYTkxODIzYTc3NTBhMjhiMGYwNjY0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.W2_v8hn-gm3nauPFp6CnvAiO4nH3FmRHa0jj5O1LKxY)

If you first delete the formula and then enter the new one, it will work! Remember that your sales price is correct as long as your total annual kWh supplied does not exceed the total annual kWh consumed! If this is exceeded, only the sales price P x 1.21 applies to the excess. This is because of the netting (salderingsregeling) arrangement.

Robbot50, 2024-04-02:
> Hi all,
>
> To make it easier to set a Dynamic ESS system up, we'd like to have a predefined list of formulas. Can you help us by sharing your formula? These are the fields we need to know:
>
> * Country [...]

Purchase and sales prices are not correct! 0.03 (fee) and 0.13 (tax on energy) are already incl. VAT.!!!!!
I also see double VAT calculations here and there in other responses!!!!
The ANWB and EnergyZero prices for 2024 are 0.0484 (fee) and 0.1316 (tax on energy) and already include VAT!!!

rattkin, 2024-04-03:
in Czech Republic we need schedule for the price calculation even with dynamic price. Because service surcharge has a time based schedule. Everybody has basically a  different schedule, because there are 3 electricity grid operators and each has a hundered time slots for each of the 5 electricity products They change the time 5 times a year for some people as well.

marcsowen, 2024-04-04:
- Germany
- Provider: Green Planet Energy
- No bidding zone
- Buy price formula: `(p*1,19)+0,1984`
- Sell price formula: fixed. Depending on date of installation.

[image]

Source: https://green-planet-energy.de/privatkunden/dynamischer-stromtarif

joaofernandes, 2024-04-09:
Is there any possibility of defining a price calculator endpoint? I'm asking this because we do have a few providers in Portugal that do quarter hourly prices with 24h ahead of time.

Robbot50, 2024-04-09:
@mpvader My opinion leaves it as it is now! Keep it simply! It is absolutely impossible to program a template for every country, region and supplier! And it can be different every year, quarter and month!

joaofernandes, 2024-04-09:
@Robbot50 I slightly disagree, if Victron could provide an extensible hook where we could easily open MR to add new configurations, the community could easily not only provide formula based configurations but also more complex solutions as well. 
Probably most of those configurations, **simple formula based approach** will solve the vast majority of use cases but, imo, having the opportunity to provide more complex calculation mechanisms should also be available. Victron is known for it's openness, this is a great opportunity to get the community contribution

Robbot50, 2024-04-09:
@joaofernandes The topic starter asks for "predefined list of formulas"! Templates for every possibility, that would be very nice, I'm looking forward to it and who knows? But suppliers come and go. Different methods and rates come and go. I don't believe in letting users add formulas! I already see several incorrect formulas passing by! For the templates I don't think Victron has time to make all the changes every day! It goes without saying that the option must be available and remain available to create the most complex calculation rules yourself. The topic started in September 2023 Unfortunately, Victron has not yet responded to many comments.

dirkjanfaber, 2024-04-09:
We only recently added a lot of pre-defined formula's to the dropdown. But it remains hard to determine which one is most correct. For instance in Germany the provider Tibber determines the formula based on postal code, which probably has to do with the distance to the power plants.
Downside from having an incorrect formula in the dropdown is that users blindly use them without checking them against their contract. So sometimes it is even better to provide none.

That being said, I'll check the proposals in this issue tomorrow and match that against what we have in right now and update the dropdown list where I can.

A flexible endpoint could indeed be a solution. I'll see if I can adjust the code to accept `msg.buy_formula` and `msg.sell_formula`, so that can be used in the Node-RED implementation.

dirkjanfaber, 2024-04-12:
FYI: Just released version 0.1.18, which allows [altering the formula's dynamically](https://github.com/victronenergy/dynamic-ess?tab=readme-ov-file#dynamically-change-formulas) by injecting `msg.buy_price_formula` and `msg.sell_price_formula`. Do note that this is a bit crude as the algorithm will use this formula for the the full schedule, making it less suitable for fixed prices (the VRM implementation handles that a lot better). But feel free to test it and provide feedback (preferably in a new issue).

joaofernandes, 2024-04-12:
> FYI: Just released version 0.1.18, which allows [altering the formula's dynamically](https://github.com/victronenergy/dynamic-ess?tab=readme-ov-file#dynamically-change-formulas) by injecting [...]

Will test it out with node-red once I get my flow to connect correctly with the VRM portal. I've enabled dynamic ESS on both setttings & within the dashboard but I'm unable o switch it to node-red mode. 
Currently getting inside node-red :"The device is no longer available via the stored service path. Please (re)select the device from the dropdown and deploy again."

dirkjanfaber, 2024-04-13:
@joaofernandes  You can only switch to the Node-RED mode from Node-RED, by injecting a `4` into service `com.victronenergy.settings`, path `/Settings/DynamicEss/Mode`.  
In order to troubleshoot the message "The device is no longer available via the stored service path. Please (re)select the device from the dropdown and deploy again." I will need some more info (flow / screenshots). Can you file a new issue for that?

joaofernandes, 2024-04-13:
I'll try to inject the value but currently the dropdown is readonly and no option is available. I'm also getting a could not read properties, which might lead to this state. 

If unable to set the mode to 4 (nodered mode) I'll log a bug

magnus-jo, 2024-04-16:
Sweden
Bidding zone (some countries have that, like Sweden and Italy).
Energy provider Greenely Premium AB [](https://greenely.com/) 

Buy price formula: p*1.25+abs(0.0014+0.0741+0.0438+0.0864)
Sell price formula: p*1.25+abs(0.05+0.6)

Elcertifikat 0.14 öre/kWh
Avgift för balansansvar 7.41 öre/kWh
Ursprungsgarantier (fossilfritt) 4.38 öre/kWh
Omkostnader - Elinköp  8.64 öre/kWh

Påslag för såld el 5.00 öre/kWh
Skatte återbäring 60.00 öre/kWh

mrjjomen, 2024-04-23:
Finland we have Nordpool market price anounced around 3pm for next day.
Price to pay when buying electricity is as follows:
(Spot price + margin) x VAT so e.g. (0,10 + 0.0039)x1,24 = 0,128836€/kWh
Margin varies based on selling company about 0.39-0.90 c/kWh
When selling electricity back to the grid it is as follows:
(Spot price - margin)
Margin can similar to buying margin or it can be without margin so just spot price as it is.

patatman, 2024-04-23:
I'm seeing a price difference on the Zonneplan formula.
I have it set the same as my Home Assistant Entso-e formala (`{% set s = {"handling_fee": 0.01653, "energie_belasting": 0.10880} %} {{(current_price + s.handling_fee + s.energie_belasting ) | float}}`)

Which results in the following formula:
```
The Netherlands
Provider: Zonneplan
No bidding zone
Buy price formula: (p+0.01653+0.10880)*1.21
Sell price formula: (p-0.01653+0.10880)*1.21
```

The pricing in DESS now matches that of the app.

Robbot50, 2024-06-12:
Hi EricCoen, yes that is possible, but in the VRM portal under Dynamic ESS settings there is an option to enter the battery costs separately.

Robbot50, 2024-06-12:
> > **Energy supplier EnergyZero (ditto ANWB) for NL**
> > Prices per kWh for 1 hour
> > 2024 Buy costs = 0.0484 Please note this is already incl. VAT
> > 2024 Energy tax = 0.1316 Please note this is already incl. VAT
> > 2024 Epex hourly price = P excl. VAT
> > 2024 VAT = 1.21
> > For the Dutch Energy tax (price in the list is excl. VAT !!) at see:
> > https://www.belastingdienst.nl/wps/wcm/connect/bldcontentnl/belastingdienst/zakelijk/overige_belastingen/belastingen_op_milieugrondslag/tarieven_milieubelastingen/tabellen_tarieven_milieubelastingen
> > Buy price : (P x 1.21) + 0.0484 + 0.1316
> > Sell price : (P x 1.21)
> > Sell price without “saldering” (refund purchasing costs and energy tax)
> > Note: Sell price (P x 1.21) + (0.0484 + 0.1316) can also be used but with the following side note! Selling price incl. discount, (refund of purchasing costs and energy tax per kWh), based on “Salderingsregeling”) can be applied if one is sure that the total number of kWh feed-in on an annual basis does not exceed the number of kWh of feed-out (delevery). If there is more feed-in then for the difference (= surplus) is the sell price is: P x VAT
> > Unfortunately, this condition cannot be included in the current DESS formula!
> > In my opinion, the ANWB formulas mentioned above and the current ANWB formula in the DESS setting are not correct for 2024, due to double VAT calculation.
> > I look forward to all reactions.
> 
> What about the energy efficiency of the Victron and solar devices? Should we add "* 0.85" to buy price formule?

Hi ErikCoen, yes that is possible, but in the VRM portal under Dynamic ESS settings there is an option to enter the battery costs separately

ErikCoen, 2024-06-13:
> Hi ErikCoen, yes that is possible, but in the VRM portal under Dynamic ESS settings there is an option to enter the battery costs separately

I am a little confused because I saw a interesting info:

[image]
Q: What about the efficiency losses of the battery?
Dynamic ESS on VRM takes battery losses into account. The system looks at the dbus service com.victronenergy.settings, path Settings/DynamicEss/SystemEfficiency.
It was set to 90 (percentage?). Is it applied instead of any modification on sell/buy price formula?

huskyte, 2024-06-17:
Can someone comment on Switzerland? Here we have sort of mixed bag:
Every quarter the the prices are fixed based on the mean day ahead pricing based on the quarter. So, while this is probably easiest coverd by a fixed pricing, it needs updating every 3 months. On the other hand, the help page claims DESS pricing to be implemented? Or is that only for those that specifically gone market?
I know, this calculation is sort of only an approximation and I don't really understand the legislative calculation rules, but is someone willing to look into that?

JeroenKetels, 2024-07-21:
**Belgium**
**Provider**: ENECO
**Area** (IMEWO - Normal)
Buy price formula: (100/8760)+(0.102*p+1)*1.06+4.72+(15.14/365)+(41.77/35040)+(1.55+5.0329+0.2042)+(182.51/2920)
Sell price formula: (100/8760)+(0.1*p-1.188)+0.472+(15.14/365)+(4.177/35040)+1.55+5.0329+0.2042+(182.51/2920)

The calculation is propable very open for interpretation.
does this breakdown of the formule seem correct to you according to this document:

https://cdn.eneco.be/downloads/nl/general/tk/BC_032_012407_NL_ENECO_POWER_DYNAMIC.pdf?_gl=1*pqdc36*_up*MQ..*_ga*MTcyNTE2NzQwMi4xNzIxNTM4ODEw*_ga_QGTX2B6GX4*MTcyMTUzODgwOS4xLjAuMTcyMTUzODgwOS4wLjAuMA..&

### The formula:
**(100/8760)**
> this is the fix cost per year so I have divided by 8760 quarters  (because SMR3 is a quarterly time meting)?

**(0.102xp+1)x1.06**
> this is the calulated energy cost.

**4,72**
> the Afnametarief (€cent/kWh) is an extra cost depending the area you are come from and whether you have only night tarief, in this case normal.

**(15.14\365)**
> the cost Databeheer - (€/year) is always SMR3 because otherways we cannot get a dynamic contract.

**(41.77/35040)**
> another annual cost: Capaciteitstarief - (€/kW/jaar)
> so again this fix cost is devided by number of quarters in a year (because SMR3 is a quarterly time meting)

**(1.55+5.0329+0.2042)**
> additional cost for green power (€cent/kWh)
> + another special tax
> + additional energy contribution

**(182.51/12)**
> contrbution to another energyfonds by month (2920 = 35040 querters in a year divided by 12)

Robbot50, 2024-07-22:
> **Belgium** **Provider**: ENECO **Area** (IMEWO - Normal) Buy price formula: (100/8760)+(0.102*p+1)_1.06+4.72+(15.14/365)+(41.77/35040)+(1.55+5.0329+0.2042)+(182.51/2920) Sell price formula: [...]

Dear Jeroen, unfortunately your formula is not correct. 
You can never include the various fixed annual or period costs in a kWh price!! 
kWh prices and fixed costs are two different things and can never be combined!
Simply because the number of kWh is variable over a period (year), and fixed prices are really fixed per year! 

The SMR3 meter has four separate registers (counters) for the grand totals of the four tariff periods.
In another register, the amount of kWh used and the fed back into the grid per quarter of an hour is recorded. 

Periodically, the grand totals and the quarter-hour measurements, including the date and times, are automatically forwarded to the grid operator, who forwards the data to your energy supplier. The four quarter-hour measurements are added together per rate type to arrive at a recording and return delivery per hour! These numbers are settled with the exchange "hourly" prices of the hour in question, including all surcharges per kWh.
The billing takes place per kWh and not per 15 minutes!!!!!!!!!

VE DESS rate formula works exclusively with the variable cost!  Energy suppliers charge per kWh and the associated variable costs and fixed costs and never per 15 minutes!

Good luck creating the right formula

[36 more replies]