victronenergy/dynamic-ess #74: [Bug]: TG anything above 1 outputs lots more KW.

KajLehtinen, 2023-09-20:
### VRM portal ID

d41243cd7f0b

### Country / region

Sweden - SE3 (se-3)

### B max

20

### TB max

5

### FB max

9

### TG max

2

### FG max

11

### Buy price

(p+0.06+0.00)*1.25

### Sell price

(p-0.00+0.01)*1.25

### feed-in possible

yes

### feed-in possible

yes

### Version

0.1.5

### What happened?

When setting TG variable to anything other than 1 the systems starts to feed in to grid over 3 Kw totally. 

Tried setting TG to 2, 3, 1.5 and nothing worked, it was always selling more to grid than allowed. When setting it back to 1 it worked, now its steady at -640 W, even though the setpoint is -1000 W.

3 Phase install

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-20:
Can you try running the latest candidate Venus software (3.20~4) and check if that resolves this?

KajLehtinen, 2023-09-20:
A sorry, didn't think of that - I even read the release notes for 3.20~4 :-(. 
Will try, the next window for selling is currently tomorrow during the day so cant verify until then.

KajLehtinen, 2023-09-20:
One question, should the value under ESS in the Cerbo change when the TG max is changed? Currently it seems fixed to -1000 but that might be my Homey control loop that left it like that when I disabled that for testing D-ESS.

KajLehtinen, 2023-10-01:
3.20~4 fixed the problem so this can be closed.

closed 2023-10-01 by KajLehtinen