victronenergy/dynamic-ess #75: [Feature]: Min Soc for selling to grid

KajLehtinen, 2023-09-20:
### Contact Details

kaj261@gmail.com

### Description

Hi!

It would be nice if a option to define 'min soc to sell to grid' was implemented, that way the system can be configured to only sell to Grid when the batteries is charged up, and it gives the option to the operator of the system to define different values for summer and winter. 

/Kaj 

### Suggested Solution

.

fofo444, 2023-09-21:
I have a similar request. [Feature]: Planned battery SOC at evening. #70

dirkjanfaber, 2023-09-21:
Just to let you guys know that we are all seeing the features, but are not responding to them yet. 

The current focus is to get everything stable and running in VRM first. Once that is done, we will start working on adding features.

disaster123, 2023-09-21:
@dirkjanfaber i just saw that there is already a setting in the remote console:

[image]

isn't it?

dirkjanfaber, 2023-09-21:
Yes, that works too. If you prefer to do that from Node-RED, you can do that using the custom output node (copy/paste this via the import menu option in Node-RED):  
```
[{"id":"c68ae0135adcce51","type":"victron-output-custom","z":"287d3e7906396df3","service":"com.victronenergy.settings","path":"/Settings/DynamicEss/MinSoc","serviceObj":{"service":"com.victronenergy.settings","name":"com.victronenergy.settings"},"pathObj":{"path":"/Settings/DynamicEss/MinSoc","name":"/Settings/DynamicEss/MinSoc","type":"number"},"name":"","onlyChanges":false,"x":700,"y":940,"wires":[]},{"id":"f6aace05b3c8309e","type":"inject","z":"287d3e7906396df3","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"10","payloadType":"num","x":310,"y":920,"wires":[["c68ae0135adcce51"]]},{"id":"190cfcf1d5c4cc55","type":"inject","z":"287d3e7906396df3","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"20","payloadType":"num","x":310,"y":960,"wires":[["c68ae0135adcce51"]]}]
```

Combine this with a [bigtimer](https://flows.nodered.org/node/node-red-contrib-bigtimer) or so to adjust for summer or winter.

KajLehtinen, 2023-09-22:
Hi!

If setting it from within the remote console under ESS/DESS, will the Node-red implementation 'know' about the minimum DESS Soc? Currently it doesn't seem so for me atleast. 

Last night DESS planned that my install, around 10am this morning would be down to approx 25% Soc, but overnight it has consumed from grid approx 3.8 Kwh so the Soc is around 45% at 7am, and DESS changed it plans so that it will sell to grid between 8-11 am. But now when its 8.30 it still hasnt started selling which I contribute to that I set the Minimum DESS SOC in Remote Console to 50% last night.
[image]

/Kaj