victronenergy/dynamic-ess #76: [Bug]: unable to disable feed-in

disaster123, 2023-09-21:
### Contact Details

through github

### VRM portal ID

e45f019ec891

### Country / region

Germany (de)

### B max

10

### TB max

4

### FB max

4

### TG max

4

### FG max

4

### Battery costs

0.01

### Buy price

(p+0.1574)

### Sell price

0.07

### feed-in possible

None

### feed-in possible

no

### Version

0.1.5

### What happened?

A bug happened!

while feed-in is disabled dess page shows enabled feed in and it feeds in...

[image]

[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-21:
Can you update to the latest candidate release (3.20~4) of Venus os and report if that fixes it?

disaster123, 2023-09-21:
Mhm I could but I’m a little bit afraid if that works - isn‘t flashmq included again? I read many problems about it with custom extensions… is there a beta or rc for 3.11 available as well?

dirkjanfaber, 2023-09-21:
3.20~4 indeed carries flashmq again. I may have spotted something else being wrong on the API side. Let me look a bit further into that. In that case it will be possible to get that fixed without needing your system to update.

mpvader, 2023-09-21:
Where do you read about problems of flashmq wrt custom extensions?

disaster123, 2023-09-21:
> Where do you read about problems of flashmq wrt custom extensions?

There are various reports like this one out there: https://github.com/freakent/dbus-mqtt-devices/issues/55

i haven‘t verified myself if they‘re valid.

disaster123, 2023-09-21:
i haven't changed anything on my side but it now shows feed in disabled:

[image]

Might it be that changing that option does not reschedule?

dirkjanfaber, 2023-09-21:
The API was not reporting back the feed_in value correctly. This has been fixed on the API side. Also on the node side I changed the code to use the configured value instead of the value that gets returned from the API. (version `0.1.6`).

mpvader, 2023-09-21:
hi @disaster123 , you wrote:

> > Where do you read about problems of flashmq wrt custom extensions?

> There are various reports like this one out there:
> https://github.com/freakent/dbus-mqtt-devices/issues/55
> i haven‘t verified myself if they‘re valid.

I am only aware of that one, not any others. If you are let me know.

What I can see is that there is no issue related to FlashMQ. Just some misunderstandings / suspicions.

dirkjanfaber, 2023-09-21:
Closing this issue as the schedule no longer sets the wrong value. If you have new problems with the feed_in, please create a new one or re-open this one.

closed 2023-09-21 by dirkjanfaber

freakent, 2023-10-10:
> > Where do you read about problems of flashmq wrt custom extensions?
> 
> 
> 
> There are various reports like this one out there: https://github.com/freakent/dbus-mqtt-devices/issues/55
> 
> 
> 
> i haven‘t verified myself if they‘re valid.

The problem reported with dbus-Mqtt-devices was caused by an out of date PYYaml dependency NOT VenusOS 3.10. This has since been fixed. I plan to ensure dbus-Mqtt-devices works with flashmq as it is released.