victronenergy/dynamic-ess #78: [Bug]: Feed-in despite feed-in being deactivated

mrstackit, 2023-09-21:
### VRM portal ID

 c0619ab2c843

### Country / region

Germany (de)

### B max

2.2

### TB max

1.2

### FB max

1.2

### TG max

0

### FG max

10

### Battery costs

0.04

### Buy price

(p+0.02+0.13)*1.19

### Sell price

0

### feed-in possible

no

### feed-in possible

yes

### Version

0.1.6

### What happened?

After I updated Venus to 3.10 today and started Dynamic ESS Node Red and another test, I discovered that after a certain amount of time the battery is no longer being charged with the maximum possible power, but is being fed into the network. However, I deactivated feeding into the power grid in Node Red. The error is only fixed again when I set mode 0 and then mode 4 again.

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-21:
Can you also set feed_in_control_on to false? Not sure if it has to do with it, but it makes not a lot of sense to have this one enabled and the feed_in disabled.

mrstackit, 2023-09-21:
So far I have understood that feed in possible and feed-in control have different uses. The first is for feed in to grid and the second is because the battery discharge can be planned and if the prices are negative the battery is not used. Is it false?

Now i have feed-in control disabeld. I'll get back to you tomorrow at the latest to see if it's of any use

I'll get back to you tomorrow at the latest to see if it's of any use.

mrstackit, 2023-09-21:
What I noticed now that I changed it and the PV no longer produces enough is that the Dynamic ESS somehow limits the power from the battery. Watch in the picture here:

[image]

Although in the Dynamic ESS dashboard it looks as if he is not planning to make any purchases:

[image]

kyros32, 2023-09-26:
Hello, yes, I can confirm this issue via beta VRM portal. I think it is related to ESS mode being stuck.

I also think, that this problem is much wider in its philosophy and such issues are complicated to solve and debug with native tools (VRM, console, HW reboot, ...).

I will elaborate further...

Some Modbus settings change console settings (let's call them internal regulation variables - which should NEVER be allowed to change by BMS persistently).

External control should always only substitute persistent setting variables (user defined via console) and should be valid and effective ONLY when communication works OK.

However, when communication with BMS is lost, then reverting to default (fallback) internal settings should happen immediately (does not happen NOW - now last communicated values are pernament).

Therefore, in such cases we see weird behaviour, because simply turning off Modbus, rebooting, reconfiguring does not work - ie. ESS mode can not be set via console - so BMS (3rd party company) can effectively break our installation.

I suggest implementing a **heart-beat variable for Modbus**. All settings made by BMS (or higher level apps and functions) should not be persistent and after comms are lost then heart-beat variable does not change it's value (ie every 30s) and then Cerbo should fallback to internal (default user defined) values and drop everything that was substituted via BMS.

Unless this is solved, I think implementing more complex d.bus and Modbus apps will be very problematic and not transparent.

Martijnth, 2023-09-27:
I'm noticing the same problem when enabled DynESS. This morning it started buying power at a way more expensive rate then it did sell an hour before. The battery was 50% so it could've used the battery but it did not.
[image]

The energy prices differ a lot between buy/sell because I ignore the 'saldering'
[image]

So even if i have a 50% charged battery it is going to buy expensive power
[image]

dirkjanfaber, 2023-09-27:
@Martijnth what is your vrm id?

Martijnth, 2023-09-27:
@dirkjanfaber 48e7da870ed3

kyros32, 2023-09-27:
@dirkjanfaber Mr. Hoor, if you would be kind enough, I am trying to make my ESS work again - disabling DynESS does not help, I even tried changing stuck ESS mode via Modbus TCP Client - I changed it successfully, but it did not help either :-(

Any support is appreciated at this time. Otherwise, I will have to reconfigure everything from scratch and cross my fingers :-)

This site has been OK until 500 -> 506 upgrade and then enabling DynESS (I wanted to test with fresh FW...).

Thank you.
Regards,
Robert

ID: c0619ab1eb6e

[image]

dirkjanfaber, 2023-09-27:
@kyros32 :As far as I can see, your issue is not Dynamic ESS related. I suggest posting it on the community forum instead in a new post.

khostri, 2023-09-28:
> @kyros32 :As far as I can see, your issue is not Dynamic ESS related. I suggest posting it on the community forum instead in a new post.

Hi @dirkjanfaber . I want to add to this as I think it is somehow DynESS related. I have tested previous version and disabling it was ok - like deploying disabled flow and then work normally as before DynESS test. I have current version (working with SOC) and I fear that its different story. Tried like 3 days ago to disable it. Do disabled DynESS flow and changed ESS mode to charging. And nothing happened - setting was totally ignored. Then I have tried to turn off nodeRED and still no charging. So now I am stuck. DynESS work otherwise ok I think although there are still some questionable timeframes. But I dont knnow what internal values was changed which are not reverted back after disabling dynESS
So I think that behavior kyros32 described when system is ignoring standard setup after DynESS was installed is valid.
So there is maybe some bug or there are some specific steps to disable it in case its needed. I hope there are still previous configured values stored somewhere.

dirkjanfaber, 2023-09-29:
You can check the mode the system is operating in. If it is in mode '0', (zero) than it is off. 

Checking the mode your system is in can be done by using the following piece of flow:
```
[{"id":"2fd11441aaae5df8","type":"debug","z":"15669bac1d63a856","name":"dynamic ess mode","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":710,"y":1020,"wires":[]},{"id":"df9c15a37ea63581","type":"victron-input-custom","z":"15669bac1d63a856","service":"com.victronenergy.settings","path":"/Settings/DynamicEss/Mode","serviceObj":{"service":"com.victronenergy.settings","name":"com.victronenergy.settings"},"pathObj":{"path":"/Settings/DynamicEss/Mode","name":"/Settings/DynamicEss/Mode","type":"number"},"name":"","onlyChanges":false,"x":370,"y":1020,"wires":[["2fd11441aaae5df8"]]}]
```

kyros32, 2023-10-01:
@dirkjanfaber @khostri Hello, as for my problem, only resolution is sticking to this manual:
https://www.victronenergy.com/media/pg/Cerbo-S_GX/en/reset-to-factory-defaults-procedure.html

1) Do Cerbo GX reset described above.
2) Set everything up again
3) Works like a charm - no need to reconfigure inverters over VE.Bus
4) After re-enabling DynESS this problem has not occurred yet - maybe fixed or it's a problem with initial set-up

So I really believe this should be looked into before going into production. Definitely some left-over persistent settings that blocks ESS.

[image]

kyros32, 2023-10-01:
@dirkjanfaber @khostri update:

now, when I set DynESS to Auto, it behaves the wrong way like before. However, when I set to Off then ESS starts working again normally. Definitely worth investigating.

-Regards,Robert

mrstackit, 2023-10-02:
Yesterday I updated to beta firmware v3.20~7 since then it seems to be working better. What surprises me a little is that the battery charge is limited and feed into the grid. As a result, the battery is not full. Why is that? I have a Pylontech US2200C, what should the battery capacity be set to 2.2 or 2.4 kWh? Maybe that has something to do with it?

khostri, 2023-10-03:
@dirkjanfaber Thanks for flow - its ok for check. I will try to test if mode=0 works.
But regarding bug feeding when should not is there in my instalation also:
[image]
[image]
[image]

disabling feedin manually (or through flow) does not help:
[image]

[image]

dirkjanfaber, 2023-10-10:
Looking back at the data of the original post of this issue (as there is quite a lot of noise in this issue).

The system calculates every few minutes how quickly it should (dis)charge to reach the target SOC at the end of the hour. If it falls behind (because your load is higher), it will speed up a bit, if it gets there too fast, it will slow down. If the battery capacity is overestimated, it will typically slow down towards the end. If the battery capacity is underestimated, it will typically speed up towards the end.
Battery manufacturers are not always exact in the battery capacity. You may have to tweak it a bit.

If you stop and start the dynamic ess, it forces a re-calculation.

closed 2023-10-10 by dirkjanfaber