victronenergy/dynamic-ess #79: [Bug]: start to feed the battery from grid instead of waiting for pv / solar power

disaster123, 2023-09-21:
### Contact Details

through github

### VRM portal ID

e45f019ec891

### Country / region

Germany (de)

### B max

10

### TB max

4

### FB max

4

### TG max

4

### FG max

4

### Battery costs

0.01

### Buy price

(p+0.1574)

### Sell price

0.07

### feed-in possible

no

### feed-in control

no

### Version

0.1.6

### What happened?

A bug happened!

See the attached pictures. dynamic ess started to feed / load the battery from grid instead of waiting for PV / solar power.

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

Marous23, 2023-09-21:
Same here. Even I am in specific time on higher % then scheduled „Targeted Soc“ system is against logic feeding the battery from grid.

dirkjanfaber, 2023-09-21:
It seems there is less PV than forecasted so DESS plans to use the grid to meet the target SOC. So this is not a bug.
The forecasts will become more accurate once the system runs longer. At the moment your system only reported the data for a week.

closed 2023-09-21 by dirkjanfaber

disaster123, 2023-09-21:
@dirkjanfaber mhm might be but the main problem seems that the wheather is not like it is reported. It's cloudy and rainy but the system says sunny:

[image]

dirkjanfaber, 2023-09-22:
The weather widget on the dashboard comes from a weather API, not solcast and no interdependence with DESS. So there is no linkage between those two.
But we are looking into why that widget was not matching reality on your system.