victronenergy/dynamic-ess #80: [Bug]: Dynamic ESS charges from grid instead of PV only

idstein, 2023-09-21:
### Contact Details

paul@strawder.info

### VRM portal ID

dca6329bad96

### Country / region

Germany (de)

### B max

15.5152

### TB max

4.2

### FB max

3.36

### TG max

15

### FG max

28.5

### Battery costs

0.04

### Buy price

0.359

### Sell price

p*0.97

### feed-in possible

yes

### feed-in control

yes

### Version

0.1.6

### What happened?

Charge / Discharge control produces additional cost when based on SoC targets only.

Given a DESS plan based on SoCs it can not adapt in the morning and evening hours properly to varying energy solar production.

See attached screenshot how the DESS continues to charge the battery (default grid set point is 0 to avoid grid consumption due to higher prices).

[image]

DESS schedule
[image]

Additionally, there is no adjustment when the predicted SoC consumption per hour deviates. For instance over night there seems to be a little bit higher consumption then predicted, but the DESS schedule forces a const lower discharge instead of switching to controlling via grid set point 0.

Without DESS grid consumption can be really minimised
[image]

When using DESS often in the morning there is an additional load due to the sharp rises of production. Maybe there should be an additional backoff battery reserve definition to deal with variances in the prediction.
[image]
 

VenuOS is running on version 3.20~2.

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

disaster123, 2023-09-21:
This seems to be a duplicate of https://github.com/victronenergy/dynamic-ess/issues/79

dirkjanfaber, 2023-09-22:
It looks like you have the minsoc set to zero. Can you try increasing that to 10%?

wetty, 2023-09-22:
Hi
your Sell Price
p*0.97
ist correct?

Normally for newer system it is less then 10 cent in Germany.

idstein, 2023-09-22:
For sell price my understand was that `p` expresses the current EPEX spot price. The 0.97 should translate into a 3% fee subtracted from the spot price.

Added now 10% MinSoC in ESS settings and will monitor the system. Consumption data should be available for almost half a year in VRM. @dirkjanfaber which data needs to be built-up?

wetty, 2023-09-22:
Sell Price is the price you get. Einspeisevergütung.

Or do you do sell direct? "Direkt Vermarktung" like Jörg from meintechblog.de

idstein, 2023-09-22:
Sell Price is either a constant value like `0.0671` like the EEG-Einspeisevergütung or a variable sell price as for my case with `Marktprämien Direktvermarktung` often linked to EPEX Spot price `p`

dirkjanfaber, 2023-09-27:
We keep the battery stable and not the grid: in our internal testing keeping the battery usage stable (basically the same as the plan) has been a lot more cost-effective than otherwise, because of the battery costs that pile up. 

In order to forecast consumption well, it needs 28 days of consumption data. Solar forecasting should be correctly functioning within a day after setting the location.

Closing this issue now as I believe the original "bug" has been explained by the way the system works.

closed 2023-09-27 by dirkjanfaber

pahorpet, 2023-10-01:
Hi @dirkjanfaber 
thank you for taking care of the issue,
I would like to add also a note to it as I see some issue regarding planing the consuption as well, seen here Dynamic ESS - d41243d31a1a (CZ)

As there is a battery cost setting, I would expect that it should react on that and if battery cost is set to 0 or close to 0 like 0.001, there should be no cost to "pile up". 

So hold grid to defined setpoint "0" or "-10" would make sence or?
I have seen that is buying some energy from grid never less there is enought battery (> SOC lim ) and not use it and the consuption is fully from grid. As the energy forecast was not accurate and it stick to the plan anyway.

DESS not allow to the grid as spot <0
[image]

Selling anyway !!!
[image]

ESS set manually to not allow to sell.
[image]

Is there any way to change the behavior to ignore SOC value and stop selling?
Is there any way to stop charging from the grid if price is higher that sell in next few hours as the plan for consuption is not possible to guess for longer period.
Is there any way to stop charging from the grid if solar forecast is higher then expected consuption?

Thank you