victronenergy/dynamic-ess #81: [Bug]: System flipping between charging /discharging forth and back from the grid within seconds each time

SimonYoungtree, 2023-09-22:
### Contact Details

siempoot@gmail.com

### VRM portal ID

c0619ab31cc4

### Country / region

Netherlands (nl)

### B max

40

### TB max

6

### FB max

6

### TG max

16

### FG max

16

### Battery costs

0.027

### Buy price

(p+0.13)*1.21+0.02

### Sell price

(p+0.13)*1.21-0.02

### feed-in possible

yes

### feed-in control

None

### Version

0.1.6

### What happened?

I see the system DESS switching from charging from the grid to discharging to the grid from ~2K to +2K and vice versa for several minutes already. (all around 8:40am 22-9-23). At the same same battery is being charged/discharged with more or less the same value.

There were no changes in the loads (changes in the graphs are cause because this is a calculated value, being temporary wrong when big changes happen).

The two screenshots give an impression of the 2 states the system was flipping.

[image]
[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-22:
It _could_ be caused by the fact that the system does not have the battery capacity set correctly, so I'd like to know if your flow contains the following piece:
[image]

If not, description if adding it can be found [here](https://github.com/victronenergy/dynamic-ess#the-battery-discharges-too-slow-to-keep-up-with-the-target-soc).

SimonYoungtree, 2023-09-22:
The issue with battery size not being set correctly was already tackled in the beginning (0.1.1 -> 0.1.3.). At that time I had slow charging discharging. This is no longer the case

dirkjanfaber, 2023-09-22:
Thanks for the info, we are looking further into this.

dirkjanfaber, 2023-09-22:
This is probably caused by the inability to hit exactly zero current at battery idle, and an SOC indication that's right on the edge. So the slightest bit of current going back into the battery sends it back to 21% and makes it go back into discharge. 
We'd like to test a fix for this (increasing the used hysteris) on your system. If you enable remote support, we can make that change for you and see if this is indeed the fix that needs to be done.

SimonYoungtree, 2023-09-22:
It has been enabled

dirkjanfaber, 2023-09-22:
Thanks. At the moment we are testing the fix on our own systems. Once we are sure that no side-effects arise, we'll put it on your system. Will let you know once that is done and when you can disable the remote support again.

dirkjanfaber, 2023-09-22:
Your system has been patched with this fix. You can disable remote support again. We are scheduling this patch to be included in the next release as well. 
Closing this issue now; please re-open the issue (or open a new one) in case the problem comes back.

closed 2023-09-22 by dirkjanfaber

SimonYoungtree, 2023-09-22:
Thanks very much. Keep up the good work. 
I have closed the remote support again and I will keep an eye on it.