victronenergy/dynamic-ess #83: [Bug]: Setpoint completely ignored in new DynESS

MichaKersloot, 2023-09-23:
### VRM portal ID

c0619ab410ae

### Country / region

Netherlands (nl)

### B max

8.5

### TB max

3

### FB max

3

### TG max

4.1

### FG max

4.1

### Buy price

(p+0.02118+0.15246)*1.21

### Sell price

(p-0.02118+0.15246)*1.21

### feed-in possible

yes

### feed-in control

yes

### Version

0.1.6

### What happened?

The new SOC method seems to completely overrule the setpoint settings.

As my solar setup connected to an easysolar ii 5kw is able to produce more Amps than I feel comfortable to send to my batteries for a longer time, I used the grid setpoint to redirect some power to the grid on high input. But at the moment this setpoint seems to be ignored completely with the 'new' SOC based implementation.

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-09-25:
How did you implement that redirection? Did you adjust the flow to top the setpoint off or so?
You can still lower the 'tb' setting.

MichaKersloot, 2023-09-25:
Hi, yes I adjust the setpoint with a separate Node-Red flow. It seems the tb setting doesn't apply to the current going from the built-in MPPT to the battery. 
Although I've got no idea how it technically works to lower the amps going into the battery and feeding the excess into the grid, while everything is on the same busbar, but hé that's your trick ;-)

JakobGode, 2023-09-26:
Maybe it is possible to use the setpoint from venus os in ess settings if dyness set the setpoint to 0.

We want to kill most grid consumption so we set the setpoint to -70W.

MichaKersloot, 2023-09-28:
You would say that, but the whole setpoint  setting seems to be ignored. F.e. at this moment, the setpoint is set to 0, the target SOC is set to 84 % and the system decides to import like 600W from the grid to top-up the 1700W Solar for charging the battery to get to this 84%. This all seems to happen, completely ignoring the Setpoint setting (and maybe other settings, like current limit?) from the Venus OS Console.

[image]

dirkjanfaber, 2023-09-28:
That is correct. When running Dynamic ESS the setpoint is ignored, as it is working with the target SOC instead. You cannot have 2 steering mechanisms working at the same time.

MichaKersloot, 2023-09-28:
Is there expected to be some knobs and/or variables to fine-tune the Dess behaviour in the future? It looks like there is some demand for that?

MichaKersloot, 2023-10-03:
Seems a won't fix?

closed 2023-10-03 by MichaKersloot

JakobGode, 2023-10-03:
@dirkjanfaber 
> That is correct. When running Dynamic ESS the setpoint is ignored, as it is working with the target SOC instead. You cannot have 2 steering mechanisms working at the same time.

Hey i think for most german users it would be nice to change the system behind dyness from target soc to setpoint.
In my case we have changing selling prices each hour and a constant consumption price.

Apparently it doesn't make sense to work with a target SOC, as the battery will then often be charged from the grid, which incurs high costs. (This is only permitted in Germany for calibration, otherwise it is illegal.)

It would be better if a program determined when the battery is charged and when it is discharged depending on the yield forecast, the consumption forecast, the sales price per hour and the system losses and battery costs. It is important that, if possible, no electricity is drawn from the grid, unless the previous sale from the battery was even better after deducting all losses. This would make DESS much more economical.

I think working with target soc may be easier but now it`s not interessting for me to use DESS beacuse auf much more grid consumption and no economical behavior.