victronenergy/dynamic-ess #84: [VRM - Bug]: Scheduling fixed prices not only changes per hour

dirkjanfaber, 2023-09-25:
### Country / region

None

### feed-in possible

None

### feed-in control

None

### What happened?

> The DESS setup page in the beta VRM portal only seem to allow me setup times using hours, not minutes. So for example I can't set it up to have fixed pricing between 23:30 and 05:30, I have to round up/ down to 00:00 and 05:00.

Reported via https://community.victronenergy.com/articles/232720/dynamic-ess-on-beta-vrm.html?childToView=233170#comment-233170 by _sieade245_

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

Stroller1024, 2023-09-26:
@dirkjanfaber This is correct.  UK requirements for 'Smart Meters', are that they can see the energy used over every half-hour period.

ollyro, 2023-10-13:
I was about to log the same ticket then found this one. I'm also in the UK. Same issue when setting up a fixed prices schedule. I'm not aware of any UK tariff or pricing scheme with a granularity of less then 30 mins.

BigBassettHound, 2024-12-19:
For CZ it would be nice to have option to add interval by minutes as low/high price is changing based on defined minutes intervals ie: High - 13:16-14:10. 
Another option is to read times from distributors api - but I haven't study it in detail yet.

ajnadox, 2025-01-31:
I would also like to possibility to have by minutes, Sweden and most of the EU will from 2025 start using 15 min reading instead of reading usage per hour.
So a change to this will be needed.

The EU info meeting about this https://www.entsoe.eu/network_codes/cacm/implementation/mcsc/#market-coupling-consultative-group-mccg